betaflight/betaflight #8466: Visual hand over of control in GPS rescue mode

butterSnake, 2019-06-24:
when the failsafe GPS rescue hands control back to the user ( when you get your radio link back ) there is no warning of the event. I reckon an osd warning with a ‘hand over in..... 3,2,1’ would be a good addition as the pilot is likely to be quite stressed at this point..  speaking to Mike on slack he thinks a maunal hand over where you have to cycle your rescue switch is also a good idea.. I like the auto hand over but a maunual hand over is likely the best as a default set up..

IvoFPV, 2019-06-24:
I assume you are running 4.0? If you are, GPS rescue hand over by stick deflection has been implemented and will be released with version 4.1

butterSnake, 2019-06-24:
Yeah I am. I think a notification on the osd is a good idea though and pretty easy to implement

IvoFPV, 2019-06-25:
Well if you regain control by moving the sticks above certain threshold like it is implemented now(not in 4.0), I think there should be no delay there, therefore no time for a osd warning message.

IvoFPV, 2019-07-16:
@RustyToad #7936

RustyToad, 2019-07-17:
Thanks Ivo. Glad to see a nice solution to this is in the pipework. I've only had a genuine unexpected failsafe once and the task saturation is very real - I had to stop my fingers hitting the disarm switch instead of the failsafe switch until it came safely back in range.

closed 2019-08-23 by stale[bot]