betaflight/betaflight #8467: Dynamic Throttle Limiting

skyfpv, 2019-06-24:
**The "Problem"**
Our quads spin the motors ~25% faster at the beginning of the pack than at the end. For me, this is a bit annoying during racing, and I imagine during freestyle sessions too. This problem is often exacerbated by time trial tracks like the MGP Global Qualifier, where consistency is extremely important. When I first take off, I'm often struggling not to clip the tops of gates, and towards the end I start bumping the ground when I unexpectedly induce voltage sag. I would really like the ability to start out my run with access to less throttle, and then have access to all the power back towards the end of the pack when I need it again.

**How Dynamic Throttle Limit Could Help**
I propose we add a "dynamic throttle limit." The user defines an initial throttle limit for the quad to start out with (say 75%). Then, as the pack drains, the throttle limit is raised to eventually be 100%. The value of the throttle limit would ideally be based on something like battery voltage, mAh used. It might be nice if this was user defined also. This could really improve (throttle/RPM) consistency, and would likely be greatly appreciated by many pilots. 

**Additional Thoughts**
Perhaps there are more intelligent ways of using the information available to the FC to determine what the limit value should be during the flight. For example, perhaps motor RPM could be incorporated somehow instead of using voltage or mAh? Also I suspect that using raw voltage might be too jumpy and so would need to be smoothed a bit before use in defining the current throttle limit. Curious what your guys' thoughts are!

etracer65, 2019-06-24:
This issue has been discussed multiple times and abandoned as unworkable. Battery voltage varies too much during flight to be useful in this regard. Also it creates a feedback loop spins out of control since by increasing throttle in response to low battery you put more load on the battery. Which then causes the battery to sag even more, causing more throttle. And so on.

Battery consumption is not a good measure of voltage output (which controls motor power). Different batteries will have significantly different curves for consumption vs. voltage.

RipperDrone, 2019-06-24:
Maybe you can define a time-dependent throttle scaling on your RC? E.g. starting at 75% throttle limiting, then increase scaling factor over 2minutes (or whatever a race time duration is) proportionately with timer value up to a max value of 100%... to exactly simulate battery drain, it might get a little more complex (non-linear correction factor for scaling)...

skyfpv, 2019-06-24:
@etracer65 I see that it has been discussed in #7099, though was not implemented. Additionally, I notice that the unstable voltage feedback was discussed already also, but no one had yet tried a longer time constant. In fact, I believe that was the last state of the issue. As @RipperDrone pointed out, I think this could work acceptably well even if it was simply time based. I think we'd like to avoid having to do it on the radio side as you'll lose a significant amount of throttle resolution

butterSnake, 2019-06-25:
Get your self some 6s action.. unless your tying to run too smaller pack you’ll have consistent cell voltage and performance till the end of the pack.. where you will literally fall out of the sky when it’s empty 😂..  besides the increases in over all performance it’s the best part of 6s

skyfpv, 2019-06-25:
[image]
Flight Video: https://youtu.be/3_Ko74LjyqU
@butterSnake I'll delete my comment if you delete yours. I'm not here to start a flame war. I'm here to make software better.

butterSnake, 2019-06-26:
You empltied the pack.. nothing you can do about that except use a bigger one or go slower.. maybe you could implement  it using the mah counter. It would need your current sensor to be actually calibrated though. I can’t see any serious racer wanting the throttle limited though as most would rather win a race than be nice to a battery

skyfpv, 2019-06-26:
The issue I'd like to focus on is not my flying ability, setups, or flight time. You seem to think these are the issue, and as the video shows that's not true. The focus of this issue is getting the throttle/RPM ratio to deviate less via a dynamic throttle limit. Obviously it will always deviate due to physical limitations of motors and batteries, but the software should at least be able to mitigate the degree to which that occurs

theSkonk, 2019-06-26:
i seem to think? if i thought any thing of the sort i'd most certainly say so... as i did say though you could use the maH counter to implement such an idea with a user set limit at which the limiting starts to happen. issue of course is it makes no account of the state of the users batteries or over all setup.. still i don't see it as a very sensible function as i do mostly race and wouldn't want to loose a race thanks to my quad limiting my throttle for me

butterSnake, 2019-06-26:
i seem to think? if i thought any thing of the sort i'd most certainly say so... as i did say though you could use the maH counter to implement such an idea with a user set limit at which the limiting starts to happen. issue of course is it makes no account of the state of the users batteries or over all setup.. still i don't see it as a very sensible function as i do mostly race and wouldn't want to loose a race thanks to my quad limiting my throttle for me

skyfpv, 2019-06-26:
@butterSnake I think you are right that going off mAh usage is the most simple/useful method at this point. A combination of mAh and voltage could be used also, possibly allowing the user influence how much the voltage is allowed to influence the throttle limiting. I recorded some data from [this](https://www.youtube.com/watch?v=f3WvJYGOLLE) flight I found and started playing with the graphs. 

We can prove that the deviation between throttle output and throttle input is theoretically lower if we come up with a working equation for it. This is what I've got so far...
[image]
Figure A: Two possible throttle limit curves compared to the adjusted voltage during a flight. Notice that mAh by itself tends to be extremely linear so the pilot likely won't even notice it changing. The mAh+voltage method will probably be noticeable and I think the ratio between the two would need to be user adjustable.
[image]
Figure B: Comparison of the standard deviation for no throttle limiting vs mAh based throttle limiting vs 50%mAh 50% voltage limiting. Notice that both equations yield significantly lower deviations, especially at the beginning of the pack when the voltage most rapidly changes. This should result in users not feeling the pack drain until it is depleted. Stick to plastic flying will (in theory) yield closer to the same thrust throughout the pack unless the user is abusing his batteries.

These are the equations I came up with
s = starting pack voltage
e = ending pack voltage
v = voltage
m = mAh used
t = total battery capacity (mAh)

For throttle limiting via mAh only...
throttleLimit = ((e/s)*(1-(m/t)))+(m/t)
For throttle limiting with mAh and voltage each weighted at 50%...
throttleLimit = (((((e/s)*(1-(m/t)))+(m/t)))*0.5) + (0.5*(e/s)*(s/v))

[Here](https://docs.google.com/spreadsheets/d/1uqOFPVs-vXLLCdLXG2CDDiMvxzrKMBuNSAhsC_Av5uI/edit?usp=sharing) is the Google spread sheet I used to work this out. Please excuse my terrible Excel skills.

danielschafer, 2019-06-26:
Why can we have vbat pid compensation and not vbat throttle compensation?
Both would be dependent on vbat.
For vbat pid compensation, is vbat somehow smoothed to get this working?
We could use the same algorithm to compensate throttle.

etracer65, 2019-06-26:
As I said above:
> creates a feedback loop that spins out of control since by increasing throttle in response to low battery you put more load on the battery. Which then causes the battery to sag even more, causing more throttle. And so on.

Increasing PIDs doesn't have a significant effect on battery load in the short term.

RipperDrone, 2019-06-26:
why not use the throttle-weighted time method from Taranis (=T2) as a simple and good approximation for battery SOC: integral / sum of: flight time since take-off x throttle %?

etracer65, 2019-06-26:
> why not use the throttle-weighted time method from Taranis (=T2) as a simple and good approximation for battery SOC: integral / sum of: flight time since take-off x throttle %?

It already exits - It's the Virtual Current Meter. Why don't you try it and see how (in)accurate it is...

RipperDrone, 2019-06-26:
have it on 3 of my builds where FC / ESC's don't have physical current meter, good enough for me - so I thought I'd throw it in for @skyfpv as an alternative option. Myself, I don't care about battery draining over race time at all :-)

skyfpv, 2019-06-26:
Ok so it sounds like right now it sounds like these are some options to try. Thanks for all the suggestions guys!
**raw voltage based**
- Only metric we theoretically care about.
- Probly wont work due to the negative feedback loop.

**filtered voltage based (running average or some such)**
- Based on the only metric we technically care about (voltage), but might be too latent.
- Will suffer from the same issues as raw voltage for short time constants. 
- For long time constants will fail to smooth out sudden drops in motor rpm due to voltage sag.

**Virtual Current Meter based**
- Will be very linear which will make its affect less noticeable to the pilot.
- Does a good job of reducing the overall deviation.
- Will work on hardware with no current or voltage telemetry.
- Will drift over time, causing dynamic throttle limiting not to reach 100%, or to reach 100% too early.

**mAh based**
- Will be very linear which will make its affect less noticeable to the pilot.
- Does a good job of reducing the overall deviation.
- Will completely fail to reduce deviation for temporary or sudden voltage drops. 
- Won't work well if a slightly dead battery battery is used.

**mAh/voltage mixing**
- Will respond quickly to sudden voltage drops
- Will allow user to prioritize either linear or aggressive throttle limiting
- Wont suffer from 

Anything I'm missing?

ecob81, 2019-07-03:
With 32bit ESC's and RPM telemetry enabled will it also be possible to cap the RPM like a rev limiter?

Dropsleutel, 2019-07-13:
Maybe we need a kind of "Governer Mode" as used on helis?  So throttle stick basically defines the setpoint of the rotor RPM in a feedback loop.  But then need to use esc telemetry to measure the RPM.  Also which motor RPM to use as setpoint? The average RPM, the slowest motor RPM?

closed 2019-08-19 by stale[bot]

quiljw01, 2019-10-08:
I'm all for this! please implement this feature!  Thank you for the writeup and thoughts @skyfpv !

Theoretically we could also do power base.  Set max power in betaflight.  Requirements are voltage and current sensing.