betaflight/betaflight #8495: Refactoring motor/ESC driver

jflyper, 2019-06-27:
As new generation of ESCs with new and multiple datalink options for connecting the ESCs to an FC are expected to emerge, it is probably a good time to start thinking about refactoring motor/ESC drivers.

Here are random notes that came up to my mind.

- New datalink options
High speed UART. Including LIN option.
A variant of SPI. may be good for FC to 4-in-1  or ESC integrated FC.
CANFD.
Can a common digital motor vector table handle different datalink options?

- Timer based DShot and GPIO-banging Dshot
Is DShot a PWM protocol??? Why driver sources are named `pwm_output_dshot_*`?

- Digital motor control command naming
Since "DShot" refers to a particular encoding of digital commands onto a group of pulses, protocol command names prefixed by `DSHOT_`  (`dshotCommands_e`) would become incorrect once non-Dshot datalink is deployed.

(To continue ...)

mikeller, 2019-06-27:
> "DShot" refers to a particular encoding of digital commands onto a group of pulses

Yes and no. 'Dshot' refers to physical encoding **and** a protocol specification, and it is well possible that even protocols that do not use the physical encoding of Dshot will be using the protocol, in order to be able to immediately adopt the features that Dshot has, like commands, or RPM feedback.

Maybe we need to look into separating the physical encoding from the protocol, and introduce naming to reflect this?

jflyper, 2019-06-28:
> Maybe we need to look into separating the physical encoding from the protocol, and introduce naming to reflect this?

This is actually my point in this matter.
Digital ESC Protocol DEP (May sound funny if ESC is spelled out ... 'Digital Electronic' Speed Controller lol)
Digital Speed Controller DSC
No More SHOT NMS
...

mikeller, 2019-06-29:
Since Dshot is already in common use as a term for the protocol, and since it is already used for transmission with different physical layers I think we should stick with it as the name of the protocol. This can then be easily adapted when Dshot is used over new physical interfaces, e.g 'Dshot over SPI'.

hydra, 2019-07-14:
yeah, the pwm_* files originate back from `Cleanflight` when all the motor and pwm code was extracted at the same time from whereever it was the time.  The rx code was then extracted and recently the servo code has been extracted.  What's left is motor code now, so likely the files should be renamed to `motor_*.c/h`

closed 2019-08-20 by stale[bot]