betaflight/betaflight #8508: Rush Tank Smart Audio and Matek F722se w/ Betaflight 4.0.3 Not working

Ayougo, 2019-06-29:
Setup:
Smart Audio connected to (TX) pad/UART3 on FC and (RX) pad (confirmed continuity to Rushtank VTX via smart audio pad)

Betaflight 4.0.3 for Matek F722se (mini)  

Issue:

In the OSD (googles) Smart Audio displays Statx only and "back" with no other configurable values. 

Diagnostics:

Connected TBS Pro 5v (using same wiring) and SA works!

Summation:

Smart audio control is not working with the new Rushtank VTX/Betaflight  4.0.3 on a Matek f722se board.

etracer65, 2019-06-29:
Contact the manufacture of that VTX. You already proved that smart audio is working by using a genuine TBS vtx so the other vtx likely has a buggy implementation. Ask the manufacturer when they'll provide a firmware update for their vtx to correct.

Ayougo, 2019-06-29:
Its a licensed smart audio vtx...

etracer65, 2019-06-29:
> Its a licensed smart audio vtx...

So? That doesn't mean they implemented the smart audio protocol properly. Betaflight doesn't know anything about specific vtx's - it just uses the smart audio protocol. Clearly the protocol works as you've confirmed with a real TBS vtx. So that just leaves a problem with the implementation in the other vtx.

Docteh, 2019-06-30:
Is it possible they are using smartaudio 2.1? I don't see any mentions of protocol versions on any of the item listings. Except one mentioned v1 and that seems odd. 

@Ayougo as a purchaser of a VTX can you ask either the store or manufacturer about the smart audio version? sometimes paying customers get better responses on such questions...

Ayougo, 2019-07-02:
It doesnt seem like a hardware issue. In the CLI, I can set vtx power and the vtx will accept the change but I dont have access to the VTX through the goggles. Smart audio 2.1.

closed 2019-07-02 by mikeller