betaflight/betaflight #8510: GPS rescue reset home position by mode switch

Jetrell, 2019-06-30:
As we are aware. GPS rescue returns the quad to the location where it was armed. The home location.

I recently had a situation were a few people congregated around me while I was flying. Under this situation... If a failsafe or video failure had occurred, the quad would have proceeded to descend in a populated area. 

I know this is some what hypothetical. But I think it would be of benefit, if the home location could be reset at another location some distance out from the takeoff location, by a mode switch, at the pilots discretion.

etracer65, 2019-06-30:
GPS Rescue is not “return to home” and shouldn’t be used that way. It won’t auto-land. You’re not supposed to let it return and crash but rather retake control as soon as the radio link returns.

Jetrell, 2019-06-30:
@etracer65  
I totally understand.

But this request is to make the GPS rescue safer.
As you would have read in the title post.  There are occasionally circumstances that are beyond our control... Like people turning up at the home location.. **Or video TX or camera failure.**  _You know the old saying... If it can happen,, it probably will at some point. ;-)_

Plus with many 2.5" and 3" quads using GPS rescue.  These quads a hard to see until they are right on top of you. And we obviously need visual contact to take back control.

I think that this would give another option and element of safety to pilots.  
Do you get were I'm coming from?

etracer65, 2019-06-30:
Having it on a mode switch is pointless if you've already lost contact with the quad and forgot to reset the home position.

If you are taking off from an unsafe position with people around then that's really the problem - don't do that. Instead you'd have to remember to reset the home position in this case which is something that would be easily forgotten until it's too late. Indirectly the capability already exists for this anyway. Just land somewhere else, disarm and re-arm. The home position will be reset.

Jetrell, 2019-06-30:
I agree with your points. I try to fly as far away from people as I can.

I was just think of a case when people walk up to you in the middle of a flight and ask questions.
Under this condition. I would use this function to reset the home location somewhere safe and near buy, as I flew over above that location.... in the case GPS rescue was required during that flight.

It figured it would be quick and easy, without having to land etc.

etracer65, 2019-06-30:
Or simply tell the people that walked up to move away if you experience a failsafe and have no control...

closed 2019-06-30 by Jetrell

aidaho, 2023-11-10:
I feel like this is something I would've wanted.  
I often fly near a dirt soggy field which I have no desire to walk down to, yet nevertheless it's an arguably much safer place to RTH failsafe into, in case I cannot, or elect not to, regain control. Both for people and the aircraft itself.

I perceive flying to the field first after takeoff, setting new home from hover, and then continuing with the flight as a better alternative to the possibility of landing on trees or the roof of the structure adjacent to a takeoff location.