betaflight/betaflight #8516: GPS_Rescue arming without GPS not working

SlientEmber, 2019-07-01:
Have GPS on NOXE F4 board (BTFL 4.02 Target: NOX1), BF10.4.1.
Failsafe is set to Land not GPS.
Currently testing setup on bench
For indoors, where no gps or in areas without good reception have tried setting. 

**gps_rescue_allow_arming_without_fix = ON**

Check after setting and confirmed as **On**.
But if I go out of CLI and then back into, the function has switched back to **OFF**. 
OSD also displays GPS error when trying to arm. 

Based on description, I would expect that once set it remains at that setting. 
Will hold the **ON** setting only while in CLI tab. 
Is this configuration issue or fault?

```# get gps
gps_provider = UBLOX
Allowed values: NMEA, UBLOX, MSP
Default value: NMEA

gps_sbas_mode = AUTO
Allowed values: AUTO, EGNOS, WAAS, MSAS, GAGAN

gps_auto_config = ON
Allowed values: OFF, ON

gps_auto_baud = OFF
Allowed values: OFF, ON

gps_ublox_use_galileo = OFF
Allowed values: OFF, ON

gps_set_home_point_once = OFF
Allowed values: OFF, ON

gps_rescue_angle = 32
Allowed range: 0 - 200

gps_rescue_initial_alt = 50
Allowed range: 20 - 100

gps_rescue_descent_dist = 200
Allowed range: 30 - 500

gps_rescue_ground_speed = 2000
Allowed range: 30 - 3000

gps_rescue_throttle_p = 150
Allowed range: 0 - 500

gps_rescue_throttle_i = 20
Allowed range: 0 - 500

gps_rescue_throttle_d = 50
Allowed range: 0 - 500

gps_rescue_velocity_p = 80
Allowed range: 0 - 500

gps_rescue_velocity_i = 20
Allowed range: 0 - 500

gps_rescue_velocity_d = 15
Allowed range: 0 - 500

gps_rescue_yaw_p = 40
Allowed range: 0 - 500

gps_rescue_throttle_min = 1200
Allowed range: 1000 - 2000

gps_rescue_throttle_max = 1600
Allowed range: 1000 - 2000

gps_rescue_throttle_hover = 1280
Allowed range: 1000 - 2000

gps_rescue_sanity_checks = RESCUE_SANITY_ON
Allowed values: RESCUE_SANITY_OFF, RESCUE_SANITY_ON, RESCUE_SANITY_FS_ONLY

gps_rescue_min_sats = 8
Allowed range: 5 - 50

gps_rescue_min_dth = 100
Allowed range: 50 - 1000

**gps_rescue_allow_arming_without_fix = ON**
Allowed values: OFF, ON

frsky_gps_format = 0
Allowed range: 0 - 1

osd_warn_no_gps_rescue = OFF
Allowed values: OFF, ON
Default value: ON

osd_warn_gps_rescue_disabled = OFF
Allowed values: OFF, ON
Default value: ON

osd_gps_speed_pos = 234
Allowed range: 0 - 15359

osd_gps_lon_pos = 234
Allowed range: 0 - 15359

osd_gps_lat_pos = 234
Allowed range: 0 - 15359

osd_gps_sats_pos = 234
Allowed range: 0 - 15359```

After CLI tab, _diff_ has this:

```set acc_calibration = 50,-32,-119
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = ONESHOT125
set motor_pwm_rate = 480
set failsafe_throttle = 1200
set failsafe_procedure = AUTO-LAND
set current_meter = ADC
set battery_meter = ADC
set small_angle = 75
set gps_provider = UBLOX
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 30
set osd_vbat_pos = 2467
set osd_rssi_pos = 2083
set osd_tim_1_pos = 2102
set osd_flymode_pos = 2488
set osd_mah_drawn_pos = 2435
set osd_home_dir_pos = 2093
set osd_home_dist_pos = 2095
set osd_altitude_pos = 2114
set osd_stat_tim_1 = ON
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON```

McGiverGim, 2019-07-01:
Have you made a `save` command after entering the other one?

atomiclama, 2019-07-01:
Working fine for me. Can't replicate your problem.

mikeller, 2019-07-01:
@SlientEmber: Please provide the information you were asked to provide when opening this issue.

closed 2019-07-02 by no-response[bot]