betaflight/betaflight #8522: Motors are enabled before `pwmEnableMotors` is called from `init`

jflyper, 2019-07-01:
A boolean flag `pwmMotorsEnabled` (through `pwmAreMotorsEnabled`) in `pwm_output.c` is used to control the enable state of the motor control to write motor values to motors.
https://github.com/betaflight/betaflight/blob/3936bbaf253d03b8b389afe6297b5d41c5e4a49e/src/main/drivers/pwm_output.c#L248-L251
https://github.com/betaflight/betaflight/blob/aba49b39ae2ff51e84e19e36d4267b9bb5009bbb/src/main/flight/mixer.c#L479-L494

To enable this flag, `pwmEnableMotors` is called at the end of `init`, which is reasonable.
https://github.com/betaflight/betaflight/blob/aba49b39ae2ff51e84e19e36d4267b9bb5009bbb/src/main/fc/init.c#L913-L915
However, the `pwmMotorsEnabled` is, upon successful initialization, set to `true` at the end of `motorDevInit` , which is called from the midst of the `init`.
https://github.com/betaflight/betaflight/blob/8140504f844e5e380fac8ab1d3119ff50bfa51bf/src/main/drivers/pwm_output.c#L426
https://github.com/betaflight/betaflight/blob/aba49b39ae2ff51e84e19e36d4267b9bb5009bbb/src/main/fc/init.c#L534

It is most likely that setting `pwmMotorsEnabled` at the end of the `motorDevInit` is incorrect.

Thoughts?

mikeller, 2019-08-01:
@jflyper: Has this been addressed in your recent motor refactoring?

mikeller, 2019-08-31:
Pinning this to make sure it doesn't slip.

closed 2026-09-26 by blckmn