betaflight/betaflight #8523: pwmShutdownPulsesForAllMotors does not work for Dshot

jflyper, 2019-07-01:
`pwmShutdownPulsesForAllMotors` called from `pwmDisableMotors` and `stopPwmAllMotors` directly manipulates CCR register of relevant timer channels through `pwmOutputport_s.ccr`.
https://github.com/betaflight/betaflight/blob/aba49b39ae2ff51e84e19e36d4267b9bb5009bbb/src/main/drivers/pwm_output.c#L226-L234
However, the `ccr` is not initialized and remain zero for digital protocols.

- It may be possible under current Dshot implementation that uses timer channels to driver motor signals.
- For forthcoming new non-timer based motor protocols/datalinks, the implementation of "motor stop/disable" should be made for each protocols/datalinks.

Thoughts?

mikeller, 2019-07-01:
One thought is that disabling the motor output signal (even if done correctly) does not seem to be the right thing to be done as part of `stopPwmAllMotors()` - the expectation would be that the motors are stopped by continually sending the '0 speed' value in whatever protocol is used.

jflyper, 2019-07-02:
I think it's the semantics of the word '_shutdown_'. To me, '_shutdown_' is beyond 'commanding zero-speed', but deactivating a system, or moving the system to another, more primitive state. Under analog PWM protocol, writing zero to CCR causes signal to go flat, and I believe it is 'no signal' for ESCs.

mikeller, 2019-07-02:
I guess `pwmShutdownPulsesForAllMotors()` is an ambiguos name for a function these days:
- in PWM protocols, 'no signal' gets the motors to disarm, so 'shut down pulses to disarm' makes sense;
- in Dshot, disarming happens already when a 0 Dshot value is sent - shutting down pulses causes the ESC to de-initialise and start scanning for PWM / Dshot signal again, so shutting down pulses does not make sense.

jflyper, 2019-07-03:
I think we'd need to clarify several things.

- I believe BLHeli manual suggests '_armed_' state as receiving valid signal, regardless of protocol used. This might be different from how we use the term.
- BLHeli has '_reset_' state, but not '_shutdown_'. Are these equivalent?
- There are several motor functions named after '_shutdown_' and '_stop_'. Are we differentiating these two?

mikeller, 2019-07-03:
> I believe BLHeli manual suggests 'armed' state as receiving valid signal, regardless of protocol used. This might be different from how we use the term.

I guess if the ESC is controlled by a flight controller and not directly by an RC channel like in manually controlled fixed wing aircraft then the concept of 'arming' ESCs is also a lot less relevant.
The Dshot specification declares a '0' value to mean 'disarmed' - not sure what that is supposed to mean in terms of ESC behaviour though.

> There are several motor functions named after 'shutdown' and 'stop'. Are we differentiating these two?

Isn't 'shutdown' used when the signal going to the ESCs is disabled, and 'stop' just means 'stop motor'?

> BF use 'idle', too. What exactly is this?

'idle' is (or at least should be) used to indicate the output signal required to spin the motors at the speed they have when armed and at minimum throttle, without influence from the PID controller.

> How values like min_command and digital_idle_percent related to above states?

They all mean different things, except that there is overlap between `dshot_idle_value` and `min_throttle`, as both are used to define the motor speed at idle, for different protocols.

jflyper, 2019-07-04:
@ronlix 
I need your help.
How does an BLHeli ESC behave in each mode when it sees the "no-signal" condition on the signal line?

- All PWM modes (legacy, *Shot)
- DShot mode

And in DShot mode, are the "no-signal" condition and the "zero-value" condition different?

ronlix, 2019-07-04:
I cant speek for blheli directly, but the normal way is:
- 0 signal (dshot:0, pwm:1000us) means armed, but no rotation.
- no signal (input permanent high or low) means disarm after a few ms, waiting for a valid 0 signal.

On kiss and fettec escs a permanent high signal for abaut 0,5s let them reset to their bootloader.

jflyper, 2019-07-04:
@ronlix Thanks. (And sorry for my confusion with 4712)

@4712 How about BLHeli?

sskaug, 2019-07-04:
It's basically the same as for kiss. No signal (low or high line, or invalid dshot frame) for a given time (a few hundred ms) causes signal to be considered lost.

jflyper, 2019-07-04:
I found another interesting code regarding motor stop/shutdown.

Hard fault handlers call `stopMotors` which writes 0-signal to motors instead of shutting them down. I wonder if this is a correct action, as the handler should do its best to force the craft to fall from the sky.

Thoughts?

mikeller, 2019-07-07:
@jflyper: Not sure what is better there. Stopping the motors (as fast as possible) is probably better than the 'disarming' that some ESCs implement (i.e. continual rebooting, leaving the motors to free-wheel).

jflyper, 2019-07-15:
Okay, here's the current call graph for motor stop / shutdown functions.
I wonder if all call patterns are sensible...

[Betaflight call graph for motor stop shutdown.pdf](https://github.com/betaflight/betaflight/files/3390890/Betaflight.call.graph.for.motor.stop.shutdown.pdf)
[image]

mikeller, 2019-07-15:
Thanks for creating that, @jflyper. At least for `cliDshotProg()` and `cliSerialPassthrough()` but most likely also for `esc4wayInit()` and `escEnablePassthrough()`, `pwmDisableMotors()` can be replaced with `stopPwmAllMotors()`, leaving us in a situation that has only two choices to stop motors:
- stop them turning but keep them armed;
- stop them turning and then shut down the signals sent to them.

closed 2019-08-22 by stale[bot]