betaflight/betaflight #8531: firmware flash fidelity. no issue, abject curiousity

alto777, 2019-07-02:
I have many times loaded firmware and flashed it down to the flight control board, comforted usually by the feedback (all good) and the "verify" phase.

I ask here what the probability is of everything looking fine, and passing whatever constitutes verification, yet the bits on the FC are not the bits in the firmware object.

I am not looking at this as a source of any trouble. I have the same problems everyone has A to Z and this isn't even on my checklist. I just wonder how close to probability 0.0 of "silent" failure we all can assume this process to have.

All I am sure of is that it is way closer to 0 than 1.0. I say again there are 999 things I'd look into first.

TIA

alto777

etracer65, 2019-07-02:
The firmware is written and then a verify pass is made to read it back and compare to the local copy. Success is only reported of the verify step passes.

alto777, 2019-07-03:
Yes, let's hope so. I understand that. What I am asking is if/how the process can appear successful, yet end up with bad friwmare on the FC.

I can think of several ways this might happen, albeit of expremely low probability. Perhaps one would say impossible… just trying to get an idea of the level of confidence y'all place in this. As I stated, it is not where I am looking for trouble, no would I.

a7

etracer65, 2019-07-03:
We feel to process is more that adequate- particularly in the context of hobby-grade electronics and open source projects.

If you have specific issues or concerns then raise them. If you want further assurances the feel free to examine the source code.

closed 2019-08-09 by stale[bot]