betaflight/betaflight #8535: Function pointer: Check for NULL before calling or call empty function?

jflyper, 2019-07-04:
This is a generic question on how to code function pointer calls, in terms of execution efficiency.

Suppose we have a function pointer
```
void (*func)(int a);
```
which may point to some valid function, or a function that does nothing like
```
void nullFunc(int a)
{
    // Empty
}
```

Now, there are two ways to make a wrapper function for a call through this function pointer.
One way is to call func straight, provided the pointer is set to `nullFunc` when no valid function is to be called.
```
void wrapperStraightCall(int a)
{
    (*func)(a);
}
```
Another way is to check for a NULL pointer before calling, provided the pointer is set to NULL when no valid function is to be called.
```
void wrapperWithCheck(int a)
{
    if (func) {
        (*func)(a);
    }
}
```

My impression is
- `wrapperWithCheck` is better if NULL function pointer is the dominant or critical path case; it does not execute any calls if the pointer is NULL; savings on call execution, stack operation and cache churning.
- `wrapperStraightCall` is better if non-NULL function pointer is the dominant or critical path case; savings on one comparison and one conditional branch.

Comments?

etracer65, 2019-07-04:
I'm not sure there's one "right" answer and that it depends on context. If the common case is that the function is not defined/used then I would lean towards the `wrapperWithCheck` method. But the context and knock-on effects are important. Take for example the filtering chain where `nullFilterApply` is used for filters that aren't active. Changing to the `wrapperWithCheck` method would probably cause the speed optimized code to explode in size with all the conditional paths as the functions get inlined.

jflyper, 2019-07-04:
Ah, I wasn't aware of the inlined ones. Good info.

mikeller, 2019-07-04:
@etracer65:

> Changing to the wrapperWithCheck method would probably cause the speed optimized code to explode in size with all the conditional paths as the functions get inlined.

Not sure about the impact of this - all it is is a simple null check per vtable function call - function calls through a vtable obviously can not be inlined, as there is no way for the compiler linker to know what function will be called.

But I think there the correct answer will be a case by case:
1. in cases where performance matters and not calling a function is acceptable, null checks are preferable to 'placeholder' functions, as it avoids the costly, non-inlinable function call. In terms of code size, the extra checks will probably come in smaller than the extra function, unless there is a lot of them;
2. in other cases where not calling a function is acceptable, both are acceptable, 'placeholder' functions resulting in a bit cleaner, better maintainable code. But could / should probably be folded into 1. for simplicity;
3. in cases where not calling a function is an error condition the hard fault from calling null is the desired outcome.

ledvinap, 2019-07-08:
IMO it's more about coding style than performance.

With NULL check it's aditionall `cbz` instruction (1 cycle + 2 cycle pipeline reload if branch is taken). Before calling the function, caller-saved registers has to be stashed somewhere, this may be done only in conditional branch. This should not be significant if register pressure is small (compiler can use callee-saved registers).
After function call, all globally-visible memory is considered modified (there is almost no way to tell C compiler about clobers caused by function except `pure` and `const` attributes). Speed optimization may avoid some reload when NULL branch is taken (duplicating code) ... 
From quick test, it seems that gcc with `-O3` is happily duplicating the function body, testing the function pointer on function entry. This will considerably increase code size.

With placeholder nil function, there is register call overhead (1 cycle + 2 cycle, same as conditional branch) and `BX LR` (again 3 cycles). Dummy function in only 2 bytes long (unless value is returned), maybe 4 with alignment, so it has about same impact as test.

@mikeller : 
1. `-O3` may considerably increase code size (duplicating and unrolling function)
3. IMO there is no explicit protection to cause hardfault when jump to NULL is taken. It will end up executing vector table, probably triggering invalid instruction eventually ...

mikeller, 2019-07-08:
@ledvinap: So gcc actually inlines all functions that are assigned to function pointers anywhere in code? Also, I am surprised that the function calling overhead (compared to the NULL check approach) is only so small.
I guess 3. is more of a theoretical case anyway, function pointers that require a callee function to be present should always have one assigned.

etracer65, 2019-07-08:
@mikeller My comments relating to the use case:
```
void wrapperWithCheck(int a)
{
    if (func) {
        (*func)(a);
    }
}
```
were that for speed optimized code all calls to this are likely to be inlined and the resulting condition would result in the code exploding in size due to unrolling and all the different code paths.

Whereas the case we're using now in filtering (`nullFunction`) won't cause this. So it's really about avoiding the conditional in my opinion.

mikeller, 2019-07-08:
@etracer65: Do you mean the calls to `wrapperWithCheck()` being unrolled, or the calls to `*func()`?

etracer65, 2019-07-08:
`wrapperWithCheck()`. I'd think the function is going to be inlined and the resulting condition in the code will cause the function calling it to be unrolled.

mikeller, 2019-07-08:
@etracer65: The only extra cost for inlining `wrapperWithCheck()` would be the NULL check - my assumption was that calling `*func()` would not be inlined, as the compiler can not know the address that `func` will be pointing to, and thus incur the cost of a non-inlined function call. I wasn't aware that they are so cheap on STM32.

hydra, 2019-07-14:
I prefer a solution with less logic.  The code is generally more maintainable.  You never even have to think about *why* the function might not be present in every case you see the `if`.

mikeller, 2019-07-14:
'maintainable' is a stretchable term - in my opinion code that is easier to read (i.e. shows locally that we do not try to execute an unassigned function pointer) is easier to maintain than code that is based on assumptions, but forces the reader to check every occurrence of the function pointer in the entire code base to get certainty that there are no accidental calls to `NULL()`.

closed 2019-08-20 by stale[bot]