betaflight/betaflight #8539: Use local time for blackbox log files when RTC is present (patch included)

SJChannel, 2019-07-06:
**Describe the bug**
Blackbox log files have date/time stamps in UTC, even when "timezone_offset_minutes" is non-zero.  Timestamps in FAT filesystems are supposed to be in the local timezone.

**To Reproduce**
Use any aircraft that has a valid real-time clock (e.g., a GPS module) and that has "timezone_offset_minutes" set to a non-zero value.  Cause the firmware to create a blackbox log file.  Observe that the timestamp of the file is in UTC, not local time as it should be.

**Expected behavior**
FAT filesystem time stamps should be in local time.

**Flight controller configuration**
```
# diff all

# version
# Betaflight / KAKUTEF7 (KTF7) 4.0.5 Jul  6 2019 / 11:29:33 (0d97bff42) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEF7
manufacturer_id 
mcu_id 002500463436511731393931
signature 00000000000000000e000000000000000000000000000000ad4722007d472200

# name
name Bando Basher

# resources
resource MOTOR 5 NONE
resource CAMERA_CONTROL 1 C09

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -AIRMODE
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -RC_SMOOTHING_INIT_FAIL

# beacon

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 3 32 115200 57600 0 115200

# led
led 0 4,7::C:12
led 1 5,7::C:12
led 2 6,7::C:12
led 3 7,7::C:12
led 4 8,7::C:12
led 5 9,7::C:12

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 900 1300 0 0
aux 4 28 0 1700 2100 0 0
aux 5 35 2 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_calibration = 67,-34,62
set align_mag = CW270FLIP
set mag_declination = 15900
set rssi_channel = 8
set blackbox_p_ratio = 64
set motor_pwm_protocol = DSHOT600
set ibata_scale = 293
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 42
set osd_cap_alarm = 1000
set osd_alt_alarm = 375
set osd_vbat_pos = 2433
set osd_rssi_pos = 2449
set osd_tim_1_pos = 397
set osd_tim_2_pos = 406
set osd_flymode_pos = 332
set osd_throttle_pos = 2424
set osd_crosshairs_pos = 2253
set osd_current_pos = 395
set osd_mah_drawn_pos = 2441
set osd_gps_speed_pos = 2104
set osd_gps_lon_pos = 2063
set osd_gps_lat_pos = 2049
set osd_gps_sats_pos = 2095
set osd_compass_bar_pos = 33
set osd_altitude_pos = 2454
set osd_warnings_pos = 2401
set osd_pit_ang_pos = 353
set osd_rol_ang_pos = 376
set osd_nheading_pos = 2081
set osd_nvario_pos = 402
set osd_rtc_date_time_pos = 65
set osd_stat_rtc_date_time = ON
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set debug_mode = GYRO_SCALED
set vtx_power = 3
set vtx_low_power_disarm = ON
set vcd_video_system = NTSC
set camera_control_key_delay = 125
set timezone_offset_minutes = -420

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 111
set pitch_rc_rate = 86
set yaw_rc_rate = 74
set roll_srate = 73
set pitch_srate = 73
set yaw_srate = 73

# rateprofile
rateprofile 1

set roll_rc_rate = 61
set pitch_rc_rate = 49
set yaw_rc_rate = 54
set roll_srate = 73
set pitch_srate = 65
set yaw_srate = 40

# rateprofile
rateprofile 2

set thr_mid = 33
set roll_rc_rate = 61
set pitch_rc_rate = 49
set yaw_rc_rate = 54
set roll_srate = 77
set pitch_srate = 71
set yaw_srate = 49
set tpa_rate = 80
set tpa_breakpoint = 1800

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

**Setup / Versions**
 - Kakute F7 flight controller (target KAKUTEF7) -- but this bug applies to all targets.
 - I tested with the 4.0.x maintenance branch, but the bug is present in master as well.

**Additional context**
See the attached patch file for a fix.
[fat_local_time.patch.txt](https://github.com/betaflight/betaflight/files/3365028/fat_local_time.patch.txt)

etracer65, 2019-07-06:
The FAT filesystem doesn't actually specify any rules for the timezone used for file creation dates. While it's true that Windows generally treats them as "local time", other operating systems like Linux tend to treat them as UTC.

SJChannel, 2019-07-06:
For what it's worth, Mac OS treats FAT timestamps as local time, even though its system clock is UTC.  Since the FAT filesystem is defined by Windows, I think Betaflight should treat it the same way Windows does.

Docteh, 2019-07-09:
how come you didn't submit a pull request?

SJChannel, 2019-07-09:
Sorry, I am brand new to the GitHub development model.

Docteh, 2019-07-09:
ah, fair enough. I thought linux usually tries to match linux when doing filesystems.

SJChannel, 2019-07-09:
I did a few experiments on various types of operating systems, and here's what I found:

- Windows treats FAT timestamps as being in the local timezone
- Mac OS treats FAT timestamps as being in the local timezone
- Linux (Ubuntu 18.04.2 LTS) treats FAT timestamps as UTC (but see below)
- FreeBSD treats FAT timestamps as UTC

Microsoft invented FAT16 and FAT32, and they specify in lots of places that the timestamps are in the local timezone.  See, for example, this article:

https://docs.microsoft.com/en-us/windows/win32/sysinfo/file-times

Interestingly, the Ubuntu man page for MOUNT(8) lists the following mount options for FAT filesystems:

```
       tz=UTC This  option disables the conversion of timestamps between local
              time (as used by Windows on  FAT)  and  UTC  (which  Linux  uses
              internally).   This is particularly useful when mounting devices
              (like digital cameras) that are set to UTC in order to avoid the
              pitfalls of local time.

       time_offset=minutes
              Set  offset for conversion of timestamps from local time used by
              FAT to UTC.  I.e., minutes will be subtracted  from  each  time-
              stamp  to  convert  it to UTC used internally by Linux.  This is
              useful when the time zone set in the kernel via  settimeofday(2)
              is  not  the  time  zone used by the filesystem.  Note that this
              option still does not provide correct time stamps in  all  cases
              in presence of DST - time stamps in a different DST setting will
              be off by one hour.
```
I tested them, and the first option has no effect whatsoever.  It is accepted, but it doesn't do anything.  I also tried "tz=PDT" and "tz=America/Los_Angeles", but the mount command rejected them as invalid.

The second mount option ("time_offset") works as advertised.  But of course it doesn't handle daylight savings time automatically.

I still think that since FAT is a Microsoft filesystem, we ought to do what Microsoft does.  But I don't want to beat the topic to death, so I will stop now.

Docteh, 2019-07-09:
I think its worth submitting as a PR. You might get more coders looking at it.

I applied the patch and threw it into a branch. You might be able to submit that as a PR.
https://github.com/betaflight/betaflight/compare/master...Docteh:sjchannel?expand=1

I haven't been using blackbox logging so I have no opinion on the matter myself ;)

SJChannel, 2019-07-10:
Thanks, Docteh, I appreciate it!  Let me do some reading and get up to speed on how things are done around here, and then i'll submit the PR.

closed 2019-08-16 by stale[bot]