betaflight/betaflight #8550: Consolidating DMA channel/stream name/type

jflyper, 2019-07-09:
There are two types of DMA implementations used for STM32 family; stream based ones for F4, F7 and H7, and channel based ones for F3 and now G4. These two have different register type names, `DMA_Stream_TypeDef` and `DMA_Channel_TypeDef` respectively, and variables/structure members for these objects are often named `xxxDMAStream` and `xxxDMAChannel` respectively (with a rare case that two are named `ref` (which is yet another topic for discussion) in `drivers/dma.h`).

This distinction between two similar objects makes code harder to read. For example, here is a codelet from `drivers/serial_uart.c`:

mikeller, 2019-07-09:
I think 1. and 2. are pretty much equivalent in how this would be handled in the code using DMA. 1. seems to be how it is done in other places in the codebase. 3. will require a lot of disambiguation everywhere DMA is used, so it's probably not desirable.

jflyper, 2019-07-11:
That's my impression, too.

And I recalled that we have another related issue on BDMA support for H7.

Unlike F-series, H7 have both stream based and channel based DMA modules. For this reason, H7 library no longer declare pointers to DMA instances as pointer to specific types but `void *`.

https://github.com/betaflight/betaflight/blob/07621912d2589b941085c35b9eef740f07afe704/lib/main/STM32H7/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_dma.h#L153-L196

While for user codes that statically embed specific channel/stream still can use `DMA_Stream_TypeDef` and `DMA_Channel_TypeDef` typed variables in assignment to `Instance`, it is probably better for us to just use something like
```
typedef void dmaRegister_t;
```

What do you think?

ledvinap, 2019-07-11:
@jflyper : It's better to use struct for opaque data types - `struct dmaRegister_s; typedef struct dmaRegister_s dmaRegister_t`. This will provide type checking, but it's otherwise similar to `void`.

Using `typedef void` is IMO bad idea - C is relaxing some warning on void type and it will be confusing without detailed knowledge.

Or just use `void`, but without `typedef`

jflyper, 2019-07-11:
@ledvinap It's a new technique for me, but sure make sense. I'll more than consider it for the new code.

ledvinap, 2019-07-11:
@jflyper : It's quite common way to make opaque object in C ...

jflyper, 2019-07-16:
Any good name for the consolidated register object/entity?

`dmaRegister_t` is too ... naive?

`dmaStrchan_t` or `dmaStrChan_t`
`dmaEngine_t` ... too exaggerated?
`dmaChannel_t` ... preferential treatment to DMA_Channel_TypeDef? Confusion with older code?

jflyper, 2019-07-17:
Or, even

dmaType_t

lol

jflyper, 2019-07-17:
`dmaUnified_t`

mikeller, 2019-07-17:
`dmaResource`?

:bulb:

jflyper, 2019-07-17:
Now I kinda understand why variables and members are named `ref`; it's too hard to name it right 🤣

mikeller, 2019-07-17:
As they say:

There are only 3 hard problems in computer science:
- naming;
- off-by-one errors.

jflyper, 2019-07-17:
Oh, I recalled HAL libraries call them "**_instance_**", e.g.:

```
    dmaHandle.Init.Instance = DMA2_Stream5;
```

mikeller, 2019-07-17:
I think 'instance' is an appropriate name when used in a transitive way (i.e. something pointing at it as 'a particular one out of a set', like a handle saying 'this is my instance of the DMA 'thing''). As an intrinsic name I don't think it is very good, as it includes the 'one particular' connotation.

closed 2019-08-23 by stale[bot]