betaflight/betaflight #8551: GPS rescue mode: A method to change ascent speed ?

mactac, 2019-07-09:
**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

mactac, 2019-07-09:
I know that you can set return speed, however although the RTH speed is reasonable, the speed at which it initially rises to the RTH altitude is way too fast.   I'm running Li-On batteries on my long range cruiser, and the speed at which it rises will smoke a battery.

Is there some way to independently change this, or is perhaps this overly fast rise some sort of bug in the code?

RTH will basically kill my battery & drop the quad to the ground.

etracer65, 2019-07-09:
This is available in BF4.1.

closed 2019-07-09 by etracer65