betaflight/betaflight #8553: level mode not working correctly, drifting gets worse over time

wildflyer, 2019-07-09:
I have 4 quads with BF 4.0, all of them are drifting in level mode. everytime I start in level mode, it start drifting to front/right, when fly around a little it gets worse and the drift angle gets to 45+ degrees after only a few stick inputs and is not flyable anymore. I need level mode for gps rescue, otherwise gps rescue doesnt work ... any ideas ?

IvoFPV, 2019-07-10:
This is accelerometer drift and happens on some flight controllers and is not a problem with firmware itself. Sometimes it is just a worn out sensor from crashing, a manufacturing defect or what is most probable in your case a specific IMU used on those flight controllers that is affected.

etracer65, 2019-07-10:
Reduce vibration - the accelerometer is very sensitive to vibration and this will cause it to drift. Soft mount your flight controller, make sure all frame screws are tight, check for cracks/extra flex in the arms, replace/balance props, replace damaged motor bearings.

wildflyer, 2019-07-10:
ok, I checked vibrations with log file, there is almost no vibrations, everything built very stiff. I have the noxe flightcontroller from BG, is that known for problems ? I also remember some months ago level mode worked very well, but I cannot reproduce what firmware or parameters I had. it was colder outside, is temperature maybe the reason ? and is there maybe a test I could make (I can log) to determine what the cause is ?
greetings
joachim

etracer65, 2019-07-10:
The accelerometer samples too slowly and is too filtered for you to see any meaningful vibration information in the blackbox log. The vibrations are affecting the sensor internally.

You're just going to have to troubleshoot and try changing components. It's possible that swapping the flight controller may help as the accelerometer can become damaged over time due to crashes.

Closing this now as it's not a Betaflight firmware issue.

closed 2019-07-10 by etracer65

atomiclama, 2019-07-11:
@wildflyer 

>  it was colder outside, is temperature maybe the reason ?

Yep the sensor is temperature dependent. There are a few things you can do to help.

- Calibrate the ACC once the FC is at it's stable temperature. Don't do it from cold.
- Fine tune the level flight with the stick commands. I calibrated the ACC on a level bench and it would not fly level till I tweaked it.
- Cross your fingers.