betaflight/betaflight #8568: Enable vtx low power trigger for failsafe

ijustwannafly, 2019-07-12:
**Is your feature request related to a problem? Please describe.**
Betaflight still only allows vtx low power on disarm.  When failsafe occurs vtx power remains at the last high power setting until radio link and disarm is re-established.

**Describe the solution you'd like**
Vtx low power on disarm AND/OR failsafe BUT only failsafe if GPS is not being used ie if GPS is not configured then vtx low power can be triggered by failsafe as well.

**Describe alternatives you've considered**
Failsafe usually happens on radio link loss and there would be no way to reset vtx power unless you regained link.  With modules that have model finder options it is better to scan for the lost model at 25mW versus anything greater than that.

**Additional context**
This shouldn't be allowed to trigger when GPS is enabled and being used, especially for GPS rescue where failsafe is also used to trigger GPS Rescue.  In this case we wouldn't want to set the vtx power low for fear of the pilot losing video feed when the failsafe occurs and the quad is actually returning to the LZ.

SchonRichtig, 2019-07-15:
I still prefer a switch/poti solution for vtx power levels (an extra mode).
In this case you could set it to the lowest level in failsafe stage 1, when the radiolink comes back it would also switch back to your last vtx power setting.
Your also could switch your vtx power levels during flight, or build an rssi based solution on you transmitter with open tx.

ijustwannafly, 2019-07-15:
Unfortunately for me, I use almost exclusively matek vtx's with tramp protocol.  The matek vtx still has the "issue" of dropping video signal for a split second when it switches power levels.  With that, I don't change power levels mid-flight.

Jetrell, 2019-07-16:
> I still prefer a switch/poti solution for vtx power levels (an extra mode).
> In this case you could set it to the lowest level in failsafe stage 1, when the radiolink comes back it would also switch [...]

@SchonRichtig  I agree. It would be very convenient. And could be used in multiple ways with TX logical switches. 

@ijustwannafly I have heard of this issue with tramp. It sounds like a problem at their end.  
With SA, I've had no such issues while using the iNav VTX power level switching function.

jflyper, 2019-07-18:
@ijustwannafly 

Looking for a lost craft with directional VTX antenna is the last resort when everything else is disabled or non-functional. Why do you want to power it down?

yoodog, 2019-07-18:
> @ijustwannafly
> 
> Looking for a lost craft with directional VTX antenna is the last resort when everything else is disabled or non-functional. Why do you want to power it down?

I do not agree with that. I fly near forest and on windy days you can not hear ESC beaping from more than 50m. I do not like special beapers nad do not use gps - super light build. I have small 100mha bat with charge circuite to vtx and cam - this gives me abput 40min to find my quad in forest.
 Always enough and always gets me clear video and ALWAYS find my quad ;-D

switch to 25mw up on failsafe would be awesome 
dynamic power output based on feed analysis would be crazy futuristic (F7 board will become popular haha), but rssi based is not great idea - signal degradation is too exponantial. 99-95-90-60-25-0

closed 2019-08-24 by stale[bot]