betaflight/betaflight #8569: Target: IMPULSERCF3 no BLHeli Passtrough, no F.Port Telemetry on UART1

RaptorFPV, 2019-07-12:
**Describe the bug**
TARGET: IMPULSERCF3

1.BLHeli Passtrough doesn´t work on BF Versions above 4.0.0.
  Could use BLHeli Passtrough with BF 3.5.

2.No Telemetry Data on F.Port (UART1), control works.

**Flight controller configuration**
```# diff
# Betaflight / IMPULSERCF3 (IMF3) 4.0.3 Jun  1 2019 / 11:30:27 (094cfc956) MSP API: 1.41 / FEATURE CUT LEVEL 10
batch start
name Helix

feature INFLIGHT_ACC_CAL
feature TELEMETRY
map TAER1234
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 1 1700 2100 0 0
set acc_calibration = 22,31,-19
set rssi_channel = 16
set fpv_mix_degrees = 45
set serialrx_provider = FPORT
set serialrx_inverted = ON
set min_throttle = 1040
set motor_pwm_protocol = MULTISHOT
set bat_capacity = 1550
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 350
set vbat_warning_cell_voltage = 355
set yaw_motors_reversed = ON
set small_angle = 45
set pid_process_denom = 2
profile 0

set vbat_pid_gain = ON
rateprofile 0

set roll_rc_rate = 214
set pitch_rc_rate = 214
set yaw_rc_rate = 214
set roll_expo = 85
set pitch_expo = 85
set yaw_expo = 85
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0
batch end```

**Setup / Versions**
 FrSky R9 Mini Receiver (F.Port enabled) connected to Helix FC (IMPULSERCF3) via UART1.

**Additional context**
Same Setup works perfectly fine with Motolab Cyclone v1.2 (F3) FC.

When using command: set serialrx_halfduplex = ON 
The controls don´t react to my inputs anymore and no telemetry data either. 

Control only (without Telemetry) works with: 
set serialrx_inverted = ON and set serial_halfduplex = OFF.

Thank you very much! 
Sincerly, Raptor_FPV

mikeller, 2019-07-13:
@RaptorFPV:

> 1.BLHeli Passtrough doesn´t work on BF Versions above 4.0.0.
> Could use BLHeli Passtrough with BF 3.5.

BLHeli passthrough support has had to be removed from this target in 4.0 in order to make the firmware fit into flash.

> When using command: set serialrx_halfduplex = ON 
> The controls don´t react to my inputs anymore and no telemetry data either. 

> Control only (without Telemetry) works with: 
> set serialrx_inverted = ON and set serial_halfduplex = OFF.

This suggests that your RX is connected to the RX pin of UART1. It needs to be connected to the TX pin for FPort to work, and then you need to set

```
set serialrx_inverted = ON
set serial_halfduplex = ON
```

RaptorFPV, 2019-07-13:
Thank you very much for the quick Information! 
I will connect from the RX pin to the TX pin and try again.
After succes I'll give feedback and this case is closed!

Thanks again and keep up the great work!

RaptorFPV, 2019-07-13:
Connected and tested as advised, Telemetry now works like a charm!
Thank you very much again!

closed 2019-07-13 by RaptorFPV