betaflight/betaflight #8571: esc telemetry mamba f722 +506 esc

robschii, 2019-07-13:
**Describe the bug**
esc telemtry  with mamba f722 + mamba 506 esc not working 
voltage is 18,1V and schows 5S but i only have 4s connected
esc current sensor show 0 Amps

**To Reproduce**
rx6 set to esc sensor
esc telemtry on

**Expected behavior**
i hobe it is a bug 
tell we when it is not featured

**Flight controller configuration**
# feature

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 0 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 3 0 19200 57600 0 115200
serial 4 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1800 2100 0 0
aux 1 13 2 900 1200 0 0
aux 2 26 2 1800 2100 0 0
aux 3 30 0 1300 1700 0 0
aux 4 35 0 900 1300 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set dyn_notch_min_hz = 80
set acc_hardware = NONE
set min_check = 1020
set max_check = 2000
set rssi_channel = 8
set rc_smoothing_type = INTERPOLATION
set rc_smoothing_input_type = PT1
set serialrx_provider = CRSF
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set vbat_warning_cell_voltage = 345
set vbat_scale = 107
set beeper_dshot_beacon_tone = 4
set small_angle = 180
set pid_process_denom = 1
set osd_rssi_alarm = 40
set osd_vbat_pos = 2465
set osd_rssi_pos = 2082
set osd_tim_1_pos = 2519
set osd_tim_2_pos = 2487
set osd_flymode_pos = 2508
set osd_throttle_pos = 2455
set osd_vtx_channel_pos = 2103
set osd_crosshairs_pos = 200
set osd_ah_sbar_pos = 200
set osd_ah_pos = 200
set osd_current_pos = 2432
set osd_mah_drawn_pos = 2402
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 375
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2153
set osd_avg_cell_voltage_pos = 2497
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2346
set osd_nheading_pos = 311
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_power = 2
set vtx_freq = 5658

# profile
profile 0

set dyn_lpf_dterm_min_hz = 80
set dyn_lpf_dterm_max_hz = 175
set anti_gravity_gain = 10000
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set pidsum_limit = 1000
set throttle_boost = 0
set p_pitch = 60
set d_pitch = 55
set f_pitch = 90
set p_roll = 65
set d_roll = 55
set f_roll = 90
set p_yaw = 52
set d_min_roll = 45
set d_min_pitch = 45
set d_min_boost_gain = 45
set d_min_advance = 0

# rateprofile
rateprofile 0

set roll_rc_rate = 88
set pitch_rc_rate = 88
set yaw_rc_rate = 88
set roll_expo = 35
set pitch_expo = 35
set yaw_expo = 30
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 80
set tpa_rate = 10
set tpa_breakpoint = 1650
set tpa_mode = PD

# end the command batch
batch end

**Setup / Versions**
diatone mamba f722 + mamba 506
crsf
runcam
unify race

etracer65, 2019-07-13:
Well to start with, you don't even have ESC telemetry enabled in the `diff` you provided - No UART is configured and the feature `ESC_SENSOR` is not enabled. Secondly the voltage is provided by the ESC firmware. If it's not accurate then you need to calibrate it in the BLHeli32 firmware.

robschii, 2019-07-14:
ah sorry wrong diff  this diff is when i have disabeld it again.   voltage is readinng wron okay this i can calibrate    but why is current not working?

mikeller, 2019-07-14:
@robschii: Not all ESCs actually have the shunts required to measure current - if voltage works on yours but current doesn't, then that is a sign that yours don't have a working current sensor.

robschii, 2019-07-14:
@mikeller  okay normal shunt works   but why is voltage reading wrong? 4s is shown as 5s

etracer65, 2019-07-14:
As I said before, for ESC telemetry the voltage is provided by the ESC. If it's inaccurate than calibrate it in the ESC. Contact BLHeli32 support for assistance.

GhostMaster9, 2019-07-16:
That ESC like all new Blheli_32 one use analog current sensor... It's not needed to calibrate anything. You can use ADC Current sensor and Vbat sensor, ask Diatone for the Scale of the Current sensor for calibration.

kuzyatron, 2019-07-26:
@robschii were you able to get the current reading?  I'm having the same issue, and would welcome any tips

closed 2019-09-01 by stale[bot]