betaflight/betaflight #8572: No I2C signal on EXF722DUAL F722

andrejpodzimek, 2019-07-14:
I tried to connect a MS5611 barometer to `EXF722DUAL`, [this model](https://www.iflight-rc.com/index.php?route=product/product&path=86_93&product_id=779). Sadly I just couldn't get any I2C signal at all. :sob: (FTR, this might have something in common with #8567, maybe.) Resources I based my attempts on:

* [MS5611 pinout on page 15](https://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Data+Sheet%7FMS5611-01BA03%7FB3%7Fpdf%7FEnglish%7FENG_DS_MS5611-01BA03_B3.pdf%7FCAT-BLPS0036): SCL is yellow, SDA is white in my setup.
  * This works with an `ANYFCF7`, so it's probably soldered just fine.
     [image]
* [STM32F722 pinout on page 49](https://www.st.com/resource/en/datasheet/stm32f722re.pdf)
  * I cross-checked all pin and pad locations involved; looks OK.
* [UART3 switchable to I2C2 on page 89](https://www.st.com/resource/en/datasheet/stm32f722re.pdf)
  * B10 / SCL2 / TX3
  * B11 / SDA2 / RX3
* [M5 and M6 switchable to I2C1 on page 88](https://www.st.com/resource/en/datasheet/stm32f722re.pdf)
  * B6 / SCL1 / M5
  * B7 / SDA1 / M6
* [Alternative UART3 location on page 90](https://www.st.com/resource/en/datasheet/stm32f722re.pdf)
  * Just to check that this is **not used**. It's not; TX3 and RX3 are indeed on B10 and B11, not on C10 and C11.

### Attempt No. 1 (I2C2 on UART3)

[image]

TL;DR: No matter what I did, TX3 and RX3 showed **nothing** on the oscilloscope, simply zero all the time. I rebooted the board a few times with the probe on, but there was zero all the time. **More details follow:**

First I enabled the MS5611 in the target configuration like this:

```diff
diff --git a/src/main/target/EXF722DUAL/target.h b/src/main/target/EXF722DUAL/target.h
index df180c906..eb939314d 100644
--- a/src/main/target/EXF722DUAL/target.h
+++ b/src/main/target/EXF722DUAL/target.h
@@ -117,8 +117,8 @@
 #define FLASH_SPI_INSTANCE      SPI3
 
 #define USE_BARO
-#define USE_BARO_BMP280
-#define DEFAULT_BARO_BMP280
+#define USE_BARO_MS5611
+#define DEFAULT_BARO_MS5611
 
 #define USE_ADC
 #define ADC_INSTANCE            ADC3
diff --git a/src/main/target/EXF722DUAL/target.mk b/src/main/target/EXF722DUAL/target.mk
index ecb89430d..5843aca2f 100644
--- a/src/main/target/EXF722DUAL/target.mk
+++ b/src/main/target/EXF722DUAL/target.mk
@@ -4,5 +4,5 @@ FEATURES    = VCP ONBOARDFLASH
 TARGET_SRC = \
             drivers/compass/compass_fake.c \
             drivers/accgyro/accgyro_spi_icm20689.c \
-            drivers/barometer/barometer_bmp280.c \
+            drivers/barometer/barometer_ms5611.c \
             drivers/max7456.c
```

Then I tried a configuration like this one (most of which was probably not needed / default):

```
resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11
set baro_hardware = MS5611
set baro_i2c_device = 2
```

Observations:

* **3.3 V supply** for the baro was OK. No shorts to ground detected anywhere.
* **Oscilloscope:** Zero on both SCL2 (TX3) and SDA2 (RX3) all the time.
* **I2C errors:**
  * 1 when `baro_i2c_device` was 2, **no matter** if the baro was connected or not.
  * 0 otherwise (with a purposefully wrong `baro_i2c_device`).
* **Boot time**: Way longer with the correct `baro_i2c_device = 2` than otherwise.
* **CLI**: Both `resource show` and `resource` list the correct I2C2 pins.

### Attempt No. 2

This was the same as Attempt No. 1, but I tried to apply [the slow I2C patch](https://github.com/betaflight/betaflight/files/3136981/4.0.1.slow_i2c.diff.txt) I've used [elsewhere](https://github.com/betaflight/betaflight/issues/4531#issuecomment-488573947), just to try if anything would change. Nothing changed, SDA and SCL were at zero all the time. But additionally, the board took ~30 seconds to boot. (!) It obviously didn't like the slow I2C hack at all.

### Attempt No. 3 (I2C1 on M5 and M6)

[image]

This was the same as the first attempt, but this time I attempted to use I2C1 on M5 and M6, i.e., I altered `target.h` further like this:

```diff
diff --git a/src/main/target/EXF722DUAL/target.h b/src/main/target/EXF722DUAL/target.h
index eb939314d..9bccf4116 100644
--- a/src/main/target/EXF722DUAL/target.h
+++ b/src/main/target/EXF722DUAL/target.h
@@ -125,10 +125,11 @@
 #define ADC3_DMA_OPT            0  // DMA 2 Stream 0 Channel 2 
 
 #define USE_I2C
-#define USE_I2C_DEVICE_2
-#define I2C_DEVICE              (I2CDEV_2)
-#define I2C2_SCL                PB10
-#define I2C2_SDA                PB11
+#define USE_I2C_DEVICE_1
+#define I2C_DEVICE              (I2CDEV_1)
+#define I2C1_SCL                PB6
+#define I2C1_SDA                PB7
+#define MS5611_I2C_INSTANCE     I2CDEV_1
 
 #define VBAT_ADC_PIN            PC1
 #define CURRENT_METER_ADC_PIN   PC2
```

Configuration (most likely useless / default anyway):

```
resource motor 5 none
resource motor 6 none
resource i2c_scl 1 b06
resource i2c_sda 1 b07
set baro_hardware = MS5611
set baro_i2c_device = 1
```

Observations:

* **Oscilloscope:** Zero on both SCL1 (M5) and SDA1 (M6) all the time &hellip; again.
  * I checked M1, M2, M3 and M4 and they had the expected signal (e.g. DShot when I switched to quad + DShot), so at least *something* does work on the board.
* **I2C errors:** Exactly as in Attempt No. 1, just with `baro_i2c_device = 1`.
* **Boot time**: Exactly as in Attempt No. 1, just with `baro_i2c_device = 1`.
* **CLI**: Both `resource show` and `resource` list the correct I2C1 pins.

Any ideas what could be wrong with I2C on this board? To solve the original problem (== **how to connect a baro**), I was also considering connecting the MS5611 to SPI instead of I2C. Betaflight can do (and does) that in some targets. But I'm not sure if there's a way to find a full SPI pinout on this board without MCU pin soldering horrors.

andrejpodzimek, 2019-07-14:
# YIPPEE :joy: :laughing:

```
# status
MCU F722 Clock=216MHz, Vref=3.30V, Core temp=59degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3060, Max available config: 16384
GYRO=ICM20689, ACC=ICM20689, BARO=MS5611
System Uptime: 165 seconds, Current Time: 2019-07-14T01:58:41.768+00:00
CPU:6%, cycle time: 126, GYRO rate: 7936, RX rate: 33, System rate: 9
Voltage: 179 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS ANGLE CLI MSP
```

TL;DR:

```c
#define USE_I2C_PULLUP
```

andrejpodzimek, 2019-07-14:
Now the only thing that leaves me worried is that the baro stops working when I touch SDA with my oscilloscope's probe. SCL is less sensitive, but sometimes also sends the I2C into a never-ending error spree. I think that's because my signal is bad. This is SCL: :scream: :scream: :scream:

[image]

What can be done here? Should I solder pull up resistors from 3.3 V to the buses? I think the built-in pull up resistors may not be designed for the overclocked I2C bus.

mikeller, 2019-07-14:
@andrejpodzimek: The built in pull up resistors are relatively weak, and more of an excuse than a useful feature - we found them to be too weak in other configurations. Try using external pull ups.

andrejpodzimek, 2019-07-14:
With 2.2 kΩ it looks slightly better, but not ideal. At least it doesn't stop working when probed. Some pulses appear to be narrow, but I still see 0 I2C errors. This is from SDA:

[image]

andrejpodzimek, 2019-07-14:
Closing this, because it (kind of) works as intended, considering that the board doesn't have any pads marked I2C.

It would be nice though to have support for more baros (only BMP280 is compiled in) and/or the internal pull ups enabled by default. Pull ups would have made this debugging easier, because there would have been a measurable signal instead of just zero.

closed 2019-07-14 by andrejpodzimek