betaflight/betaflight #8573: feed-forward automatically changed

maponarooo, 2019-07-14:
I had P vibration during flight today and analyzed the black box log.
During the analysis I found one strange thing.
Strangely, the value of the f_pitch parameter was moving very heavily during flight.
As far as I know, the f_pitch parameter value is a fixed value. is not it?
If this value fluctuates, the pitch of the gas will vibrate strongly.
This value and rc_command and set_point values all vibrate strongly even though I have not moved the pitch of the stick.
I want to know the reason.
![Image of BB]
(https://blog.naver.com/PostView.nhn?blogId=maponarooo&Redirect=View&logNo=221585272208&categoryNo=28&isAfterWrite=true&isMrblogPost=false&isHappyBeanLeverage=true&contentLength=12680#)

My FC Firmware is MATEKF405 ver. 4.0.3.
and related parameters are...

f_pitch = 50
rc_interp_ch = RPYT
rc_interp_ch = RPYT
rc_smoothing_type = FILTER
rc_smoothing_input_hz = 0
Allowed range: 0 - 255
rc_smoothing_derivative_hz = 0
rc_smoothing_debug_axis = ROLL
rc_smoothing_input_type = BIQUAD
rc_smoothing_derivative_type = BIQUAD
rc_smoothing_auto_smoothness = 10

etracer65, 2019-07-14:
If you look at the graph you'll quite clearly see that pitch RC Command is obviously changing and feedforward is responding properly to that change. If you're not moving pitch with the stick, then there's something wrong with your radio or your gimbals. Your radio is sending pitch change signals to the flight controller and it's responding exactly as it should.

maponarooo, 2019-07-15:
I would like to know How can I check radio gimbal signal spike?

etracer65, 2019-07-15:
You'll have to look for support resources related to your radio. Or possibly contact the manufacturer. This is not related to Betaflight.

closed 2019-07-15 by etracer65