betaflight/betaflight #8576: problems with mamba f722 mini (packet loss and no current sensing)

m013, 2019-07-14:
tried 4.0.4 and 4.1 #1746
blheli versions Rev32.6.4 and Rev32.6.5 respectively

the first problem - huge packet loss (up to 40%) i cant even arm (rpmfilter warning).
with Rev32.6.5 two of the motors get no data

the second - although i have telemetry readings from the esc on osd (rpm and temperature), i have no current sensing. i know you have to apply some load on motors tjo get readings, but i did it via motor test tab and still nothing. note that there is no current calibration setting in blheli suite
[dump.txt](https://github.com/betaflight/betaflight/files/3389729/dump.txt)
[image]
[image]

etracer65, 2019-07-14:
Firstly, you didn't even mention the type of ESC so it's difficult to help. But only some ESC's have the current sensor necessary to prove current data over ESC telemetry. It looks like yours don't.

Next, don't provide a `dump` - use `diff` for settings. Please provide this.

If you are using BF4.0.x you must use BLHeli32 beta version 32.6.4. If you are using BF4.1 nightlies then you must use BLHeli32 32.6.5.

Lastly, please clarify or separate your issues. You seem to be mixing different things, like trying 4.0 vs. 4.1, different BLHeli32 versions, questions about ESC telemetry for current sensor (which has nothing to do with RPM filter), etc.

m013, 2019-07-14:
@etracer65  , sorry i ve rolled back everything, cant provide the same diff
the esc is the one from the mamba stack mini 722 as i mentioned, i dont know the model, it came with the flight controller. blheli detects it as mamba 306. it has got a **curr** wire and a sensor
[image]

i followed the guide ad used different hexes accordingly ( 32.6.4 for 4.0.4, 32.6.5 for 4.1) as mentioned

the thing i copypasted

set gyro_sync_denom = 2
set pid_process_denom = 1
set motor_pwm_protocol=DSHOT600
set scheduler_optimize_rate=on
set dshot_burst=off
set dshot_bidir=on

regarding the last one - sorry, i am not an expert and dont know how the telemetry goes in case of bidir dshot, so i am not sure what is the cause

etracer65, 2019-07-14:
That ESC doesn't provide current sensor data through ESC telemetry. You have to connect the "current" wire to the current sensor input on the flight controller and use "Onboard ADC" for the current meter type (not ESC sensor).

Without a `diff` showing your settings when trying to get RPM filter working we can't help further.

etracer65, 2019-07-14:
And some flight controllers are designed to only support `dshot_burst` and turning it off results in some motors not spinning. So test that with your working non-RPM filter setup by `set dshot_burst = OFF`. If all 4 motors still spin than that's not the problem. If it makes some motors not work then sorry, RPM filtering won't work with that board at this time.

m013, 2019-07-14:
@etracer65 , yeah
with set dshot_burst = OFF only 2 motors were spinning

as for this current sensor - it is connected via  the stock connector and cables, so according to the scheme , i did everything right. 
so i guess it is not possible to tell if something is mapped incorrectly or some of the pads just lead to nowhere, right?

etracer65, 2019-07-14:
Please carefully read replies. This is wrong as you don't have current data provided over ESC telemetry.

[image]

m013, 2019-07-14:
i tried this filed. shows  0 everywhere
with onboard and esc sensors

etracer65, 2019-07-14:
Then you'll have to check your wiring or contact the hardware manufacturer for assistance.

m013, 2019-07-14:
@etracer65  , thanks i l ldo
i saw somewhere a similar problem though

closed 2019-07-14 by m013