betaflight/betaflight #8581: MATEKF405CTRL - accelerometer alignment needs CW 270 correction

madl3x, 2019-07-16:
**Describe the bug**
Betaflight 4.0.4 build for MATEKF405CTRL, the FC is mounted correctly on the frame, but the accelerometer needs to be adjusted in configuration to CW 270 degrees.

**To Reproduce**
Install BF 4.0.4 on Matek F405CTR

**Expected behavior**
Board alignment settings should not be modified if FC is mounted correctly. 

**Setup / Versions**
- Betaflight 4.0.4
- MATEK F405 CTR

**Additional context**
N/A

TheIsotopes, 2019-07-16:
Which target are you using? MatekF405 have CW270 set as default.

madl3x, 2019-07-16:
I used MATEKF405 target with BF 4.0.4 built on 1/07/2019 - 15:36.
I will as soon as I get to my drone.

jflyper, 2019-07-16:
I thought the MATEKF405 target is used for several different products with different gyro chip orientation. CTR is not what the target used as a reference.

madl3x, 2019-07-16:
@jflyper The manufacturer specifies the target as "MATEKF405" for BF and iNav. Even if it would have been shared by other boards, I doubt the original would have misconfigurations by default and to be recommended by manufacturer.

[matek documentation](http://www.mateksys.com/wp-content/uploads/2017/12/F405-CTR_1-1500x600.jpg)

Here goes the diff.
```
# diff

# version
# Betaflight / MATEKF405 (MKF4) 4.0.4 Jun 30 2019 / 14:34:05 (f3a95efa3) MSP API: 1.41

# start the command batch
batch start

board_name MATEKF405
manufacturer_id 

# name
name MADSRC

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource SERIAL_TX 11 A15

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -AIRMODE
feature MOTOR_STOP
feature GPS
feature ESC_SENSOR

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 2 115200 9600 0 115200
serial 2 2048 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led
led 0 0,0::CO:9
led 1 1,0::CO:9
led 2 2,0::CO:9

# color
color 0 0,255,255
color 1 302,255,113
color 4 60,202,255
color 6 54,0,183
color 9 190,0,193

# mode_color
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3
mode_color 6 0 9
mode_color 6 1 2
mode_color 7 0 4

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 2 1350 1650 0 0
aux 2 46 2 900 1200 0 0
aux 3 13 1 1800 2100 0 0
aux 4 28 3 900 1200 0 0
aux 5 35 3 1800 2100 0 0
aux 6 36 1 1800 2100 0 0

# adjrange
adjrange 0 0 5 900 2100 12 5 0 0
adjrange 1 0 4 900 2100 29 4 0 0

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_calibration = 71,44,-231
set baro_hardware = NONE
set rssi_channel = 16
set rc_smoothing_type = INTERPOLATION
set airmode_start_throttle_percent = 32
set min_throttle = 1070
set dshot_idle_value = 450
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1300
set current_meter = ESC
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_descent_dist = 100
set gps_rescue_allow_arming_without_fix = ON
set deadband = 5
set yaw_deadband = 8
set osd_vbat_pos = 14731
set osd_rssi_pos = 14362
set osd_tim_2_pos = 6551
set osd_remaining_time_estimate_pos = 2424
set osd_current_pos = 2432
set osd_motor_diag_pos = 58
set osd_craft_name_pos = 6156
set osd_gps_speed_pos = 2403
set osd_gps_sats_pos = 2048
set osd_home_dir_pos = 2081
set osd_home_dist_pos = 2082
set osd_warnings_pos = 14697
set osd_core_temp_pos = 2069
set osd_stat_tim_1 = ON
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set vtx_channel = 2
set vtx_freq = 5760
set vcd_video_system = NTSC

# profile
profile 0

set smart_feedforward = ON
set p_pitch = 50
set i_pitch = 50
set d_pitch = 27
set f_pitch = 70
set p_roll = 46
set i_roll = 45
set d_roll = 25
set p_yaw = 65
set i_yaw = 45

# rateprofile
rateprofile 1

set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 83
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 83
set pitch_srate = 83
set yaw_srate = 83

# end the command batch
batch end
```

jflyper, 2019-07-16:
@madl3x 
I think this is a FAQ.

https://github.com/betaflight/betaflight/issues/6761
https://github.com/betaflight/betaflight/issues/6883
https://github.com/betaflight/betaflight/issues/7121

TheIsotopes, 2019-07-16:
That's what Matek writes on her firmware page

> After flashing BF4.0, MUST set F405-STD & F405-mini first gyro to CW180 in configurator if arrow is facing forward
> 
> Keep default CW270 for F405-CTR if arrow forward

madl3x, 2019-07-17:
@TheIsotopes Ok, if that is by intent and is documented, please close the issue. 

I imagined that the target is intended to provide a default, correct configuration for each FC you write BF (not just pin configuration), for users to start their configuration with. I personally would prefer having separate targets for STD and CTR (in case of Matek's FCs) if they don't share the same settings, rather then remembering all the tweaks I need to do for a particular FC when re-flashing BF. 

Thanks for looking up to it.

etracer65, 2019-07-17:
The problem is that in this case Matek has made multiple boards - that all share the same target - but in fact have different gyros oriented in different ways. A target file can only contain one default so by definition it's guaranteed to be wrong for at least one board in this scenario.

mikeller, 2019-07-17:
@etracer65: Unified Targets provide a good solution for this - as they make the effort per target very small, it is easy to split this target up into individual boards, all with the correct gyro orientation, as is proposed in #8584.

closed 2019-08-24 by stale[bot]