betaflight/betaflight #8582: SmartAudio 2.1 doesn't work with "TBS UNIFY PRO32 HV 5G8" on BF 4.0.4

madl3x, 2019-07-16:
**Describe the bug**
I've installed Betaflight 4.0.4 on Matek F405CTR flight controller and the "TBS Unify Pro32 HV 5G8" VTX and the SA feature failed to work.

**To Reproduce**
Install Betaflight 4.0.4 on Matek F405CTR FC and setup the smart audio to work on UART3 with TBS Unify PRO32 HV 5G8.

**Expected behavior**
To be able to change channel settings and power level via OSD menu through smartaudio protocol.

**Flight controller configuration**
SmartAudio configured to work on hardware UART3, with a physical connection between TX3 and SA between FC Matek F405 and VTX TBS Unify Pro32.

**Setup / Versions**
 - Betaflight release version 4.0.4
- TBS Unify PRO32 as video transmitter
- Matek F405CTR as flight controller

**Additional context**
I replaced the VTX, keeping the exact same configuration on Betaflight, with a TBS Unify PRO HV 5G8 that doesn't use the newer SA 2.1 and the OSD menu worked.

madl3x, 2019-07-16:
What information is it missing, please?

mikeller, 2019-07-16:
@madl3x:

From what you were shown when opening this issue:

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

Mattybperth, 2019-07-17:
I have a unify pro32 hv with a cl racing fs4 flight controller running 4.0.4bf and my smartaudio 2.1 is working fine. the smartaudio wire from vtx is connected to the tx pad on the flight controller

Mattybperth, 2019-07-17:
I also set up vtx via the tbs Web site using the agent X there you can select smartaudio and update the firmware

madl3x, 2019-07-17:
@Mattybperth I haven't tried updating the firmware of TBS. I will try that. Have you been able to select the version of SA? 

Thanks for the info!

Mattybperth, 2019-07-17:
yes once I updated vtx you can choose and set it up choosing smartaudio and turning crossfire option off and power levels. since I did that it has worked for me

madl3x, 2019-07-24:
Thank you @Mattybperth. I can set up the channels, but cannot change the power levels. Is there any way in the release version of BF to set vtxtable?

Mattybperth, 2019-07-24:
you need to unlock the vtx hold the button down for 3 seconds do that 2
times then on the 3rd time hold down button again for 20 - 25 seconds then
it should be unlocked for all power settings red led should flash 3 times
check out this video on youtube time 3.19 https://youtu.be/Wzl-ELc7lF8 hope
this helps you

madl3x, 2019-07-26:
@Mattybperth I don't know how to thank you enough! I totally forgot about that. Yes, it now works beautifully.

Mattybperth, 2019-07-26:
awesome I'm glad to be able to help I'm only new to this myself and luckily
I was able to help
HAPPY FLYING

closed 2019-09-01 by stale[bot]