betaflight/betaflight #8585: Pikof4 combo innova v4 msp

sniperxnl, 2019-07-16:
I have here a pikof4 with innova v4 vtx combo, and with 3.1.7 firmware, then MSP with the True vision app is working
Uart 1 MSP 115200 activated

When i flash a diffrent firmware then the connection can not be made anymore with the True vision app.

Dump 3.1.7

> # version
> # Betaflight / PIKOF4 3.1.7 May 15 2017 / 05:25:24 (e1c4b5c)
> 
> # name
> name -
> 
> # resources
> resource BEEPER 1 A14
> resource MOTOR 1 A03
> resource MOTOR 2 B00
> resource MOTOR 3 A02
> resource MOTOR 4 B01
> resource MOTOR 5 NONE
> resource MOTOR 6 NONE
> resource MOTOR 7 NONE
> resource MOTOR 8 NONE
> resource MOTOR 9 NONE
> resource MOTOR 10 NONE
> resource MOTOR 11 NONE
> resource MOTOR 12 NONE
> resource SERVO 1 NONE
> resource SERVO 2 NONE
> resource SERVO 3 NONE
> resource SERVO 4 NONE
> resource SERVO 5 NONE
> resource SERVO 6 NONE
> resource SERVO 7 NONE
> resource SERVO 8 NONE
> resource PPM 1 NONE
> resource PWM 1 NONE
> resource PWM 2 NONE
> resource PWM 3 NONE
> resource PWM 4 NONE
> resource PWM 5 NONE
> resource PWM 6 NONE
> resource PWM 7 NONE
> resource PWM 8 NONE
> resource LED_STRIP 1 B07
> resource SERIAL_TX 1 A09
> resource SERIAL_TX 2 NONE
> resource SERIAL_TX 3 B10
> resource SERIAL_TX 4 NONE
> resource SERIAL_TX 5 NONE
> resource SERIAL_TX 6 C06
> resource SERIAL_TX 7 NONE
> resource SERIAL_TX 8 NONE
> resource SERIAL_TX 9 NONE
> resource SERIAL_TX 10 NONE
> resource SERIAL_RX 1 A10
> resource SERIAL_RX 2 NONE
> resource SERIAL_RX 3 B11
> resource SERIAL_RX 4 NONE
> resource SERIAL_RX 5 NONE
> resource SERIAL_RX 6 C07
> resource SERIAL_RX 7 NONE
> resource SERIAL_RX 8 NONE
> resource SERIAL_RX 9 NONE
> resource SERIAL_RX 10 NONE
> 
> # mixer
> mixer QUADX
> 
> mmix reset
[... 422 lines cut ...]
> set d_alt = 0
> set p_level = 50
> set i_level = 50
> set d_level = 100
> set p_vel = 55
> set i_vel = 55
> set d_vel = 75
> set level_sensitivity = 55
> set level_limit = 55
> rateprofile 0
> 
> # rateprofile
> rateprofile 0
> 
> set rc_rate = 100
> set rc_rate_yaw = 100
> set rc_expo = 0
> set rc_yaw_expo = 0
> set thr_mid = 50
> set thr_expo = 0
> set roll_srate = 70
> set pitch_srate = 70
> set yaw_srate = 70
> set tpa_rate = 10
> set tpa_breakpoint = 1650

jflyper, 2019-07-16:
What exactly is your problem?

sniperxnl, 2019-07-16:
> What exactly is your problem?

all firmwares higher then 3.1.7,then the true vision app can not connect anymore, so looks like a BF bug

sniperxnl, 2019-07-16:
3.2 is also working

jflyper, 2019-07-16:
So, I assume you are using the Innova in native MSP mode, correct?

sniperxnl, 2019-07-16:
i think so, i only activate MSP UART1 115200 thats all.

jflyper, 2019-07-16:
Do they expose source code for V4? I did the original Innova support for MWOSD. I'm not sure if they are still using their own unreleased code.

jflyper, 2019-07-16:
Well, I did a quick test using Airbot's MicroOSD v2.4 (MinimOSD upper compatible), probably flashed with MWOSD (version unknown, probably around Feb. 2017). FC is Omnibus F4 V3, flashed with 4.1.0 (dev master).

It works without an issue.

I think you should consult FFPV.

sniperxnl, 2019-07-17:
It have nothing todo with a source, it is a BF thing, otherwise BF 3.2 will also not work right?

mikeller, 2019-07-17:
The True vision app is not something that is made by us, and the manufacturer does not give us access to its internals, or even provide us with free versions for testing, so there is nothing we can do to fix it if it stopped working.
As @jflyper said, please consult with the manufacturer. Closing this.

closed 2019-07-17 by mikeller