betaflight/betaflight #8588: Displaying individual ESC voltages in OSD (3in1 ESC + individual ESC)

PhatBrett, 2019-07-18:
I have a few blheli_32 4in1 ESCs with a single broken ESC that have been converted to 3in1 ESCs.  I replace the broken ESC by placing  a single blheli_s ESC  on the arm. All the quads run great. However, betaflight OSD reports "0" for battery voltage. In betafight, the 3in1 ESC reports the correct voltage for the 3 working blheli_32 ESCs but 0V for the average. Is there any way to get the OSD to display the battery voltage of an individual ESC so I can monitor voltage or do I have to upgrade the individual ESC to blheli_32 and use its telemetry?

etracer65, 2019-07-18:
BLHeli_s doesn't have ESC telemetry and likewise the ESC hardware doesn't have voltage sensors. So it's not possible to get voltage from a BLHeli_s ESC. BLHeli32 is required.

Also flying with different types of ESCs on the same quad is not recommended as motor response is different for each ESC type and you're requiring the PID controller to continuously compensate which will reduce its availability to perform optimally.

closed 2019-07-18 by etracer65

PhatBrett, 2019-07-18:
You didn't address the question. I realize blheli_s doesn't have telemetry which is why I am trying to display an individual vbat from a blheli_32 ESC within the 4in1 ESC.  Also, mixing different ESCs on the same quad works fine.

mikeller, 2019-07-18:
@PhatBrett: No, there isn't. Betaflight does not have a way of distinguishing between a situation where only a subset of the ESCs used is capable of sending telemetry, and one where the wiring is broken and the data reported is incomplete and misleading, therefore it indicated 'faulty wiring' by not displaying the incomplete data.