betaflight/betaflight #8589: Motors do not work properly

TheIsotopes, 2019-07-18:
**Describe the bug**
since build `#1750` the motors does not work properly on my MATEKF722SE.
after arming the motors are twitching, even an increase of throttle does not make the motors spin
the motors only seem to work properly when i test it in the motortab of the configurator

back to build `#1749` and everything works normally again

imho there is something wrong with #8534 ... @jflyper pls can you check this

**Flight controller configuration**
this diff is from build `#1749` but nothing was changed in build `#1750`
```
# Betaflight / MATEKF722SE (MF7S) 4.1.0 Jul 15 2019 / 10:36:39 (f4bb75180) MSP API: 1.42

# start the command batch
batch start

# name: Mark1

# feature
feature -TELEMETRY
feature -OSD
feature LED_STRIP

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ON_USB

# map
map TAER1234

# led
led 0 9,2::AW:0
led 1 9,3::AW:0
led 2 9,4::AW:0
led 3 9,5::AW:0
led 4 6,5::AW:0
led 5 6,4::AW:0
led 6 6,3::AW:0
led 7 6,2::AW:0

# mode_color
mode_color 6 0 3
mode_color 6 1 6

# aux
aux 0 0 0 1600 2000 0 0
aux 1 13 1 1000 1200 0 0
aux 2 26 0 1800 2000 0 0
aux 3 31 0 1200 1400 0 0
aux 4 36 1 1800 2000 0 0

# adjrange
adjrange 0 0 2 1400 2000 30 2 0 0

# master
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SUMD
set blackbox_device = SPIFLASH
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1550
set use_cbat_alerts = ON
set cbat_alert_percent = 20
set vbat_duration_for_warning = 50
set vbat_duration_for_critical = 20
set pid_process_denom = 1
set ledstrip_beacon_period_ms = 350
set ledstrip_beacon_percent = 15
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900
set name = Mark1

# profile 0
set feedforward_transition = 30
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set i_pitch = 85
set i_roll = 75
set d_min_roll = 26
set d_min_pitch = 28
set d_min_boost_gain = 30
set d_min_advance = 50

# rateprofile 0
set roll_rc_rate = 135
set pitch_rc_rate = 135
set yaw_rc_rate = 120
set roll_expo = 15
set pitch_expo = 15
set roll_srate = 75
set pitch_srate = 75

# end the command batch
batch end
```
**Additional context**
a benchtest with MATEKF405 and SPRACINGH7EXTREME does not bring up the problem

jflyper, 2019-07-18:
I'll check it out tonight.

So, the issue develops only on F722 (out of your fleet), right?

TheIsotopes, 2019-07-18:
Yes, only on F722 ... F405 tested on workbench and H750 also tested in flight, all ok

using oneshot, multishot on F7 is ok ... imho only probs with dshot

jflyper, 2019-07-18:
Hmm... Tested with NERO (F722 target) and OMNIBUSF7V2 (F745 target), and both spin okay with Dshot.

Anyone else having the issue with Matek F722SE?

@TheIsotopes Do you have any other F7 board?

TheIsotopes, 2019-07-18:
@jflyper no, that's the only F7 board I have. It's the Matek F722-mini.

mikeller, 2019-07-18:
@TheIsotopes: Do you have a logic analyser, and could you check the ESC output signals for any signs of problems?

TheIsotopes, 2019-07-18:
@mikeller Unfortunately, I have no way to try this

etracer65, 2019-07-18:
I just flashed a MatekF722SE with build 1750 and put a logic analyzer on the outputs with DSHOT600. Everything looks fine. Output is very stable and no packet checksum errors.

etracer65, 2019-07-18:
Next I tested build 1750 on a quad with a MatekF722-STD (don't have a F722-SE quad, just the bare board). The F722-STD spins all the motors as expected - no problems found.

etracer65, 2019-07-18:
@TheIsotopes Try testing with a minimal flash. Just set to DSHOT600 and use the motors tab to test spinning the motors (NO PROPS!). If that works, try adding back your config in sections to see if you can replicate the problem. For example I see that you have LED strip enabled. Maybe a timer problem (but unlikely). Also you have SPIFLASH as the blackbox type so I assume you have a F722-mini (I tested with a real F722-SE).

TheIsotopes, 2019-07-18:
I have now set to defaults and changed only the motor protocol ... As I have already written above, there is no problem in the motortab, all motors are running perfekt ... and that's exactly what I can not understand

gucio1200, 2019-07-18:
I have same problem , motors are twitching before arming.

jflyper, 2019-07-19:
@gucio1200 I think you have a separate issue. The particular issue under question here is:

- Motors do not spin in DShot when armed.
- Motors do spin properly in DShot via motor tab slider.
- Motors do spin with analog protocols.

Please review your symptoms and file a new issue ticket if necessary.

jflyper, 2019-07-19:
@TheIsotopes What do you mean bye "mixed up"? Change in a particular motor slider accelerate/decelerate a non-corresponding motor???

TheIsotopes, 2019-07-19:
@jflyper Yes, the motors do not react as they should, start to twitch or stop for a moment.
With an analog signal everything is smooth, no matter how fast i move the slider.

jflyper, 2019-07-19:
@TheIsotopes Ah, no, I was asking if motor slider #1 controls motors other than #1, as you said "mixed up".

TheIsotopes, 2019-07-22:
> Ah, no, I was asking if motor slider #1 controls motors other than #1, as you said "mixed up".

no, that's all right

TheIsotopes, 2019-07-23:
@jflyper Here are two screenshots, one with dshot and the other with multishot
as you can see dshot has a lot of bad frames and on multishot everything is clean

[image]

[image]

jflyper, 2019-07-25:
@TheIsotopes The biggest issue is that I can't reproduce the issue...

Anyone having the exact same symptom on Matek F722-mini?

TheIsotopes, 2019-07-25:
that's strange ... that can not only be due to my controller

gucio1200, 2019-07-25:
I guess its more wide, I have same issues on Matek F722-mini and SucceX F7 - same behaviour - motors twitching , and even run themselves up without arm. Not fast - when they do not achieve smooth run on low throttle they spins up ad stop .

jflyper, 2019-07-25:
@gucio1200 
> On builds starting from june. I saw dshot was refactored , but it appear much before. On builds till May all works perfect smooth. 

Can you determine when it has started?

jflyper, 2019-07-26:
@TheIsotopes Can you do the same check with F4 and H7 cases (dshot only would do)?

jflyper, 2019-07-26:
@TheIsotopes @gucio1200 @McGiverGim 
[betaflight_4.1.0_MATEKF722SE-motor-refactor-fix-0.hex.zip](https://github.com/betaflight/betaflight/files/3434468/betaflight_4.1.0_MATEKF722SE-motor-refactor-fix-0.hex.zip)

TheIsotopes, 2019-07-26:
yeah jackpot 👍 I just tested your hex file
all dshot and the proshot protocol work perfectly again and show no bad frames.

hutch1001, 2019-07-27:
@jflyper. Could you write a hex file to solve this problem on the CL racing F7 (not duel). Having same issue. 4.1?

mikeller, 2019-07-27:
@hutch1001: Just grab it out of https://ci.betaflight.tech/job/Betaflight/1758/ when it's done.

hutch1001, 2019-07-27:
Dude,  

Thank you so much for the quick response. Love you guys and your work. The new shit is 🔥🔥🔥🔥🔥🔥🔥🔥!  

John Hutchinson

TheIsotopes, 2019-07-28:
fixed now in Build `#1758` and closed

closed 2019-07-28 by TheIsotopes

hutch1001, 2019-07-28:
Hover seems good will test tomorrow. Thank you guys for what you do. Will post bb log. You guys rock. 

John Hutchinson