betaflight/betaflight #8590: No gyro after flashing 4.1 stm32f4

beachbreak, 2019-07-18:
today I flashed my succex f4 whoop board (stm32F4) with BF 4.1 #1751and the gyro was no longer showing up in the configurator.  
I reflashed back to 4.0.4 (full chip erase) and problem still exists.  I only flashed, no flying in between ...newish board, no damages 
[image]

jflyper, 2019-07-18:
@beachbreak 

In addition to missing acc/gyro, your board is running on HSI, meaning the board is behaving in a strange manner.

Can you supply CLI `diff` output?

beachbreak, 2019-07-18:
Dang...I thought I took a pic of the diff....I’ll do it after work

etracer65, 2019-07-18:
Can you please be more clear about what target you actually flashed? "STM32F4" is not a target. It's sounding like you might have flashed a unified target base "STM32F411" and haven't applied a target config to it yet.

Please, no pictures of the screen. Supply the complete `diff` output as text in your reply.

beachbreak, 2019-07-19:
heres the DIFF

# version
# Betaflight / STM32F411 (S411) 4.0.3 Jun 1 2019 / 11:58:22 (094cfc956) MSP
API: 1.41

# diff

# version
# Betaflight / STM32F411 (S411) 4.0.3 Jun 1 2019 / 11:58:22 (094cfc956) MSP
API: 1.41

# start the command batch
batch start

board_name
manufacturer_id

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end

JyeSmith, 2019-07-19:
As etracer65 mentioned it looks like you havent flashed the config.  Copy and paste the below into the cli and see how you go.

https://github.com/betaflight/betaflight/blob/master/unified_targets/configs/IFLIGHT_F411_PRO.config

beachbreak, 2019-07-19:
totally worked..

jflyper, 2019-07-19:
We might need a visual sign of a board that is not given a specific config...

Docteh, 2019-07-20:
Flashed STM32F405, and did not apply config:
[image]

After applying config:
[image]

After asking for `defaults` on 4.0.4 build:
[image]

on the flight side of things, what counts as a configuration is a question that needs answering

on the configurator side of things, maybe something could replace or overlay `#sensor-status`?
[image]

mikeller, 2019-07-21:
@Docteh: Yeah, this is work in progress - manufacturers asking their users to install Unified Targets value having their custom target name showing more than their users' convenience...

cj780120, 2019-08-02:
I got the same issue!!
and I try that Copy and paste the below into the cli, than still....

please help

# diff all

# version
# Betaflight / STM32F411 (S411) 4.0.5 Aug  1 2019 / 12:28:05 (839661b20) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name 
manufacturer_id 
mcu_id 005900423437511231393434
signature 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master

# profile
profile 0

# profile
profile 1
[... 973 lines cut ...]
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 50
set tpa_breakpoint = 1500
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

#

Docteh, 2019-08-02:
@cj780120 Targets like STM32F405 and STM32F411 are bare targets that don't specify a board configuration. What board is that?

cj780120, 2019-08-03:
> @ cj780120像STM32F405和STM32F411这样的目标是没有指定电路板配置的裸目标。这是什么板？

@Docteh SucceX Mini F4 V2 Flight Tower System, thank you and good luck

Docteh, 2019-08-03:
https://github.com/betaflight/betaflight/blob/master/unified_targets/configs/IFLIGHT_F411_PRO.config 
This file is mentioned earlier in the thread, try using it

Please remove the long comment

cj780120, 2019-08-03:
> https://github.com/betaflight/betaflight/blob/master/unified_targets/configs/IFLIGHT_F411_PRO.config
> This file is mentioned earlier in the thread, try using it
> 
> Please remove the long comment

This one can work!!!

thank you!

I will try to learn more, thank you guys again.

cj780120, 2019-08-03:
@Docteh .....me again, everything look good, till I found out #1 #2 two Motors is not working.....

Docteh, 2019-08-03:
@cj780120 I tried looking up [SucceX Mini F4 V2 Flight Tower System] and I get a page that says STM32405, the link:
https://www.iflight-rc.com/index.php?route=product/product&product_id=700 

Does the chip on the board say F411 or F405? 

the detail may matter

cj780120, 2019-08-03:
https://www.iflight-rc.com/index.php?route=product/product&path=86&product_id=905

This one

Docteh, 2019-08-03:
For motor 1 and 2 , they have a document and a [hex]  that says

Motor 1 is A00

Motor 2 is B10

did they make a mistake?

I would ask them to be sure.

cj780120, 2019-08-03:
thanks a lot !  hope can get some good news!

closed 2019-09-09 by stale[bot]

SvenL03, 2022-05-20:
I just flashed with full chip erase and now it says no gyro!! I have the IFRC-IFLIGHT_BLITZ_F722 stack.
Is there a solution??

haslinghuis, 2022-05-21:
@SvenL03 what version of configurator and firmware?