betaflight/betaflight #8592: Manufacturer ID for MerakRC

MerakRC, 2019-07-20:
As per https://github.com/betaflight/betaflight/blob/master/unified_targets/docs/CreatingAUnifiedTarget.md, please assign manufacturer ID for MerakRC.

Requested Id : MERK
Name : MerakRC
URL : http://www.merakrc.com/

A target definition of MERAKF405 and  MERAKF722 will follow soon. I'm so sorry for the previous wrong operation.

mikeller, 2019-07-21:
It looks like your web page is broken, there is no contact information or anything else showing: http://www.merakrc.com/contact.html

MerakRC, 2019-07-21:
@mikeller Our website is under construction. I can write my contact information to my e-mail, and when the website is built, it will be more informative. I hope you will allow me to add Manufacturer ID first. Thanks very much.

PR #8601 (2019-07-24): Added manufacturer id for 'MerakRC'.
Fixes #8592.
closed 2019-07-24 by mikeller