betaflight/betaflight #8593: Telemetry working but VFAS shows 0.00V

bastibastel, 2019-07-20:
Hello,

I hope this is the right place to tell you my problem...

My hardware:
Vortex 150 with STM32F303
Firmware 4.0.4 IRCSYNERGYF3 01-07-2019
Taranis X7
Frsky R-XSR

My problem:
I updated from firmware 3.2.2_IRCSYNERGYF3. Before telemetry was working like it should. The X7 shows sensors like AccZ, Tmp1, VFAS, A4,...

After updating to 4.0.4 telemetry works as it should, but VFAS always shows 0.00V. 

Tryed:
Flash firmware again
Flashed firmware 3.2.2 and problem is gone
Deleted all sensors in X7 and searched ne sensors

Because I am new in this huge (and cool) universe of betaflight, I don't know, how to solve that problem. If you need more information, please tell me.

Thanks for all the nice software/firmware/...

IvoFPV, 2019-07-20:
Did you set your voltage meter sources correctly in the configurator? Do you see the voltage in your OSD?

bastibastel, 2019-07-21:
I set the voltmeter and amperemeter to "onboard", saved but after disconnecting the usb and battery and connecting again to the pc, the settings are gone.
See screenshot.
[image]

RipperDrone, 2019-07-21:
Use correct version of Betaflight Configurator, enums of elements might have changed...

bastibastel, 2019-07-21:
I use Betaflight Configurator 10.5.1.

Some more information, that might help:

- the osd in my googles says API version 1.41 and this is blinking
- I can't access the "flight control" and "pro-tuning" via osd, it says "incompatible fc API"

RipperDrone, 2019-07-21:
pls try latest official BF 4.0.4 and latest official BFC. Then enter CLI command 'defaults' reconfigure from scratch. Should do the trick.

etracer65, 2019-07-21:
The IRC OSD is a custom implementation and is not compatible with Betaflight. You'll have to get a custom version from IRC for it to work. Or possibly IRC has released an updated OSD firmware. Either way you'll need to check with them.

bastibastel, 2019-07-21:
Please check the version in the screenshot. I think, this is the latest version of BFC and BF.

[image]

I tryed "defaults" command and startet my configuration

I can select the voltage settings in BFC, save, reboot and the settings are gone, and overwritten with the old settings. 

Due to the ImmersionRC Vortex 150 is a BNF Racer there is a "TX Wizzard" when booting the first time to setup TX.
Is it possible, that the wizzard does some kind of write protection?

Just to make sure:
The error in the IRC OSD does not affect the VFAS problem?! I'll contact IRC about a new version for the OSD.

RipperDrone, 2019-07-21:
Yes, 4.0.4 and BFC 10.5.1 seems accurate. If setting voltages, saving in BFC, then they are gone, something must be wrong in the saving process? Can you try setting voltage/sensor parameters in CLI, save and then check what happens?

bastibastel, 2019-07-21:
## This is what happens, when I connect to BFC the first time after the "default" command:

# get meter
current_meter = MSP
Allowed values: NONE, ADC, VIRTUAL, ESC, MSP
Default value: VIRTUAL

battery_meter = 
###ERROR: CORRUPTED CONFIG: battery_meter = 3###
Allowed values: NONE, ADC, ESC
Default value: NONE

## this is what I set:

set battery_meter = adc
battery_meter set to ADC
set current_meter = adc
current_meter set to ADC
save

Rebooting

### This is what CLI says after reboot:

# get meter
current_meter = ADC
Allowed values: NONE, ADC, VIRTUAL, ESC, MSP
Default value: VIRTUAL

battery_meter = ADC
Allowed values: NONE, ADC, ESC
Default value: NONE

## This is what CLI says after removing all power sources (battery and usb cable)

get meter
current_meter = MSP
Allowed values: NONE, ADC, VIRTUAL, ESC, MSP
Default value: VIRTUAL

battery_meter = 
###ERROR: CORRUPTED CONFIG: battery_meter = 3###
Allowed values: NONE, ADC, ESC
Default value: NONE

Seems like my settings are overwritten by default settings, when disconnecting all power sources.

RipperDrone, 2019-07-21:
did yu enter 'default' or 'defaults'? Needs 'defaults' to revert to FC defaults in BF code...

looks like a version conflict btw what has been configured and what you are trying to overwrite / re-define. Pls mke sure o check the 'erase all' box when flashing.

RipperDrone, 2019-07-21:
reading your test squenc, there must b a problem with the FC onboard memory. It doesn't retain values after power cycling at all... re-reading after reboot is ok, but it loses memory after power cycle... :-( Guess u have to try another FC!

etracer65, 2019-07-21:
Please contact IRC.

They have implemented a layer that communicates with the flight controller over MSP that drives their custom OSD, profiles, etc. I suspect that it's not compatible with BF4.0 and it is corrupting the settings.

bastibastel, 2019-07-21:
Sorry, I ment "defaults" is what I entered.
Checked if the command worked. Everything was set to defaults.

I'll contact IRC. 

With Version 3.0.2, which I downloaded here:
[https://www.immersionrc.com/vortex-150-updates-errata/](url)
I had the VFAS in my tanaris. No matter what was in the BFC configured as battery meter and current meter.

The file name is "betaflight_3.0.2_IRCSYNERGYF3.hex".
I updated to version 4.0.4 in BFC. There is no official update from IRC. 

Beginner question:
When the Flight Controller is called "IRCSYNERGYF3" it should be compatible with my one, even though there is no official download?
Maybe there is no official download, because the osd doesen't work correctly and the battery settings are not saved correctly.

Anyway, thank you for your help. 
As always, when I have a problem, which no one else has in the huge WWW, it somehow can't be so serious.

closed 2019-08-02 by bastibastel