betaflight/betaflight #8596: CRAZYBEEF4FR - missing "dyn_filter_range" option

sebaqq, 2019-07-21:
I've tried to do some modifications in my Trashcan micro drone (based on CRAZYBEE4FR). Unfortunately some adjustments are not recognized by Betaflight. One and the most important of them is "dyn_filter_range". I've tried it on BF 3.5.7, 4.0.0 and 4.0.4 versions, and also Betaflight Configurator 10.5.1 and 10.5.0 (and also Chrome version) without success. Searched Internet and found no solution. Like I was the only one with this problem. Is there any issue with this option in Betaflight firmwares?

IvoFPV, 2019-07-21:
@sebaqq The setting to change is dyn_notch_range

sebaqq, 2019-07-22:
@IvoFPV thans a lot. I have seen this setting in the dump but wasn't sure that it is exactly what I was looking for.

closed 2019-07-22 by etracer65