betaflight/betaflight #8597: NUCLEOF446RE delay at start up

yuhahaha, 2019-07-22:
I'm trying to build a breadboard fc with NUCLEOF446RE. I uploaded the NUCLEOF446RE.hex to the board, and connected a micro usb to the OTG_DM and OTG_DP pins. However, when I try to connect to the betaflight, it takes a while for pc to recognize the usb device.

jflyper, 2019-07-22:
@yuhahaha Here, the bare NUCLEO-F446RE can be detected as virtual com port about 5~6 seconds after reset. Does yours take longer?

yuhahaha, 2019-07-22:
Thanks a bunch for your response.
No, it takes about that long. I didnt if this is normal behavior or not. Why is it taking 5~6 seconds? and how can I reduce this time?

jflyper, 2019-07-22:
> Why is it taking 5~6 seconds? and how can I reduce this time?

You found your first task.

Break at `init` and start stepping by `next` (`n`) command to see if there's any initialization function that is taking long.

closed 2019-08-28 by stale[bot]