betaflight/betaflight #8603: CRAZYBEEF4FR SPI RX doesn't support the latest ACCST D16 mode

kostanev, 2019-07-23:
**Describe the bug**
Cannot bind the internal SPI RX to the FrSky X-Lite Pro ACCST D16. The last attempt was with CRAZYBEEF4FR v4.0.4 and transmitter FW 1.1.0/1.1.2 FCC. This happens on Mobula 7 HD drone.

**To Reproduce**
Try to bind CRAZYBEEF4FR to some of the latest FrSky radios.

**Expected behavior**
I expect they both to bind like they do with my Taranis X9D Plus SE

**Flight controller configuration**
The FC configuration is as follows:
- Receiver mode: SPI RX support
- SPI Bus Receiver Provider: FRSKY_X

**Setup / Versions**
 - Flight controller - The FC HappyModel Mobula 7 HD comes with
 - Other components - RX - FC built-in SPI RX

**Additional context**
As far as I'm aware, the SPI receivers use reverse engineered FrSky firmware. Since the RX is placed/soldered internally, it uses the FC firmware to work properly. This is the reason for opening the issue here. Hope this is the correct place.

darkcody, 2019-07-29:
Same issues here attempting to bind CRAZYBEEF4FR 4.0.4 to FrSky X-Lite Pro ACCST D16

@mikeller was this just closed due to the missing missing flight controller configuration diff requested in the issue template? If so, can this issue be reopened?

My fc configuration diff is pasted below.

```
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight / CRAZYBEEF4FR (C4FR) 4.0.4 Jun 30 2019 / 14:06:48 (f3a95efa3) MSP API: 1.41

# start the command batch
batch start

board_name CRAZYBEEF4FR
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end

```

mikeller, 2019-07-29:
@darkcody: From your `diff` it looks like the RX was never bound to the TX - can you describe what you did to try and bind it?

darkcody, 2019-07-29:
@mikeller - sure:
- In Betaflight, issued the `bind_rx_spi` cli command in betaflight to enter bind mode for the rx. (I've also tried with the bind button on the board)
- Board changes from blinking lights to solid lights as expected 
- On the radio I have the internal RF set to ACCST D16 mode
- Selected the "[Bnd]" option on the radio. It then chirps indefinitely

I began by upgrading firmware on the X-Lite Pro, flashing the internal ISRM module to enable ACCST D16 mode, and then upgrading fw on the flight controller to 4.0.4. I'll also add that I'm brand new to the hobby, so apologies if I've missed something obvious!

darkcody, 2019-07-31:
@mikeller any thoughts on the above? And can this issue be reopened based on the information provided?

mikeller, 2019-07-31:
@darkcody: In your description above, after you selected "[Bnd]", did the LED on the board ever get back to blinking?

darkcody, 2019-07-31:
@mikeller no, the LEDs on the board remain solid

mikeller, 2019-08-01:
@darkcody: That's a shame - it looks like (at least) the bind sequence was changed  sufficiently by FrSky to stop the code driving the SPI attached RX from binding.
Unfortunately I do not have the X-Lite Pro, so I won't be able to debug this.

Othepic, 2019-08-05:
Hi, same here with my UR65 (CrazyBeeF3FR) ! happy to find people with the same issue...
My radio is the Taranis X9-Lite and it doesn't bind in ACCST D16 as it should
The same issue is reported with a Mobula in this video: https://www.youtube.com/watch?v=vjwn0rfGux0

turboed13b, 2019-08-08:
I can bind to both crazybeef3 and crazybee f4. Both on 3.4.1 however on the crazybee f4 it suffers from terrible range maybe 20ft or so.

0crap, 2019-08-08:
I can confirm this issue.
Recently I got a TinyHawk S (MATEKF411RX (M41R) ) which came with BF 4.0
That binds perfectly fine in D8 and D16 mode with my Taranis QX7S on OpenTX 2.2.2 (EU LBT)

After upgrading the TinyHawk S to BF 4.0.4 I can ONLY bind in D8 mode.
D16 mode is not possible, it will not bind.

0crap, 2019-08-08:
@kostanev Can you please remove "CRAZYBEEF4FR" from the topic title? Thank you.

Unco16, 2019-08-17:
Hi, I have a Mobula7 with a CRAZYBEEF4FR v2 flight controller. I was not able to bind it with a Taranis X-lite Pro, and finally found the solution thanks to this post: https://www.reddit.com/r/Multicopter/comments/cenkx4/tinyhawk_s_binding_issue/

Controller firmware: BTFL 4.0.5
Receiver provider: FRSKY_X

Steps that I did:
-Plug Mobula to usb
-Connect with BF Configuration, went to CLI and type bind_rx_spi (lights will become solid)
-From the X-Lite, select ACCST D16 and launch Bnd
-Light on Mobula are still solid but actually binding should be ok, to check it:
-type set frsky_x_rx_num The rx num should be the same one than your Taranis
-type save

turboed13b, 2019-08-17:
I have the same board. Range sucks.

darkcody, 2019-08-24:
> I have the same board. Range sucks.

@turboed13b I was having failsafe kick in as little as 20ft in front of me in D16. The range has been much better for me after switching to D8 (using the Vantac MPM Lite)

lamquad, 2019-09-09:
Hello everyone

I'm new to this hobby and it's my first post on this forum. So I'm sorry if I'm posting in the wrong place or if I fail to explain the problem with my Trashcan. I'm not too
 fluent in English, so sorry for the mistakes.

I had my Frsky Trashcan set up and running with my Taranis QX7, and I made few flights with it. I recently bought an external multi protocol module (iRange X IRX4 +) to be able to configure a URUAV UR65 DSM on Taranis. I updated Taranis with the latest OpenTX multi protocol (2.2.4) and made everything normal for URUAV UR65.

After a few days I went flying the Trashcan again and I verified that it was bind with the radio, but there was no response to the radio commands, neither to fly nor in the receiver tab of the BF. I tried to do a new bind and noticed a strange behavior with the normal bind process described in the manual, exactly as reported by other users in this thread. Trashcan begins the bind process but does not end, in which the lights on the board should flash again indicating the success of the bind.

What I already tried:

- BFC 10.4.0 and crazybeef4fr 4.06 firmware - unsuccessful
- BFC 10.5.1 and crazybeef4fr 4.06 firmware - unsuccessful
- original BFC 10.4.0 and crazybeef4fr 4.0.0 firmware - unsuccessful
(Firmware flashing with erasing chip)
All attempts were made through both the CLI and the bind button on the board.
All downloads on GITHUB.

I also tried to reset traschcan settings before trying to bind ... unsuccessful

Could any small drop cause this problem? In the first flights there were small falls, but none affected the flights.

Could there be a problem with OpenTx?

Please help me !!!

Thank you

mikeller, 2019-09-09:
@lamquad: GitHub is for discussions around _development_ of Betaflight. To get help with a configuration issue, have a look at other online resources like the Betafligtht Facebook group, or RCGroups.

darkcody, 2019-09-15:
I was finally able to get my xlite pro to bind to my crazybeef4 fc after following the instructions here:
https://www.rcgroups.com/forums/showpost.php?p=42145539&postcount=2539

> 1. Connect the FC and go to CLI.
>
> 2. Write the bind command:
> For Betaflight 4.x bind_rx_spi
> For Betaflight 3.x frsky_bind
> The Led will stop blinking
> 
> 3. Activate the bind function on the radio.
> 
> 4. Go to the CLI and write Save
> 
> 5. Stop the bind function on the radio
> 
> 6. Reboot the FC

Hope this helps someone else!

P.S. I'm currently running an opentx nightly, but I'm not sure if that has any significance for the steps above.

netizen01, 2019-09-30:
This is impacting me in a few different ways (some of them unrelated to Betaflight) and would love to see a fix. Here's a conversation I had with a NA sales rep:

" I happen to work for FrSky and our tech team has told us there is nothing in place that should stop these from binding. The frsky_x protocol was reverse engineered and added into the betaflight code. Not really part of the main development from what I understand. Our guess is that the reverse engineering was "good enough" for the old xjt transmitter but the new isrm is more sensitive to whatever the error is. But they will bind via the cli method I mentioned above. I personally have a bee brain lite and use it with an x9 lite."

mikeller, 2019-09-30:
@netizen01: The problem is that at the moment nobody of the Betaflight development team has got TX hardware capable of running the new ACCST mode - this makes it very hard to figure out what exactly was changed in the protocol that stopped this from working.

netizen01, 2019-09-30:
If I mailed someone a TX would that help? (assuming they would mail it back - I'll pay for shipping both ways) Who would that be?

famtory, 2019-10-08:
> I have the same board. Range sucks.

Same. Range problem.
and... Once disconnected, I won't be able to connect again.
To reconnect the aircraft must be reboot.

turboed13b, 2019-10-13:
I think the issue is with matek411rx boards which is what the crazybee f4 is.

I just got the new betafpv f4 fc and have the same problem. It isnt consistent either. It will fly perfectly fine one time then the next time I plug a battery in it will do the twitching deal again. Sometimes when first powering up it only "half" binds where the rssi will show up in the 40's of course It twitches and failsafes if I try to fly it like that. Seems like cycling the battery or even the transmitter will make it bind normal. 

Weird thing is I have been flying all my crazybee f3 boards without a problem. The crazybee f3 boards dont have a lna either and still get good range but the f4 boards need it to get any kind of decent range. Seems odd to me.

bjotos, 2019-10-29:
> I was finally able to get my xlite pro to bind to my crazybeef4 fc after following the instructions here:
> ...
> Hope this helps someone else!

Thanks, indeed that helped me out, but in my case I first upgraded to BF 4.1.0 (not sure if this was required) and still had no bind success.
I had to issue 'set frsky_x_rx_num=1' via CLI  and save/reboot, after that I was able to bind on first attempt (using bind button on FC).
X-Lite Pro kept on beeping, but exiting bind mode on TX made FC LEDs flash again and now I can see channels in Configurator reacting on stick commands.

I have no clue why this happens - the code uses '0' as default value for rxNum, there is a special code path for this in isValidPacket() and it should be set by value from TX in getBind1().
Maybe I'll give this a try to debug in the future (still waiting for other parts for my quad :) ).

Is there an easy way to add an additional debug console (for printfs) to the crazybeeF4/betaflight to debug such thing ?
I'm on linux, have coding/solder skills and various USB 2 serial boards laying around.

mikeller, 2019-11-02:
@bjotos: You could use what was added in #8905 for the debug output.

travot3k, 2020-01-07:
I have been having the same issue while trying to bind my Mobula7 with my Taranis QX7.

I enter bind mode via BetaFlight CLI command, or using the button on the FC, and the lights turn solid. Then enter BND mode on the TX, and nothing happens.

I have tried using D8 and D16, with no success on either. I have also tried nearly every version of BetaFlight from 3.5.7 to 4.1.1, including the default 4.0 version from the Happymodel support page.

I tried using the recommended BetaFlight Configurator version 10.4, as well as 10.6.

I have also tried using every OpenTX version from 2.2.4-2.3.4. Updated the internal RF and bootloader.

I have literally spent 2 days now, trying to get the FC to bind with my transmitter, and so far I have had no luck. So I finally decided to make an account here so that I could ask for some advice, after spending countless hours on YouTube and Google, attempting to find some info on what the problem is and how to fix it.

Can someone, anyone, please assist me with this issue? I have no idea what the problem is or how to go about fixing it. Thank you for your time.

wallace7084, 2020-01-19:
Has anyone been able to bind the trashcan with the FrSky Taranis X9D Plus SE 2019 no matter what I do it just won't bind

famtory, 2020-01-20:
firmware update above bf4.1.0.

martimbrepost, 2020-01-21:
I still have drop outs "RX LOSS", with my Tinyhawk 2 and XLite Pro.
I tried with and without telemetry.

Like Michael Keller @mikeller  said, there are still differences in the Frsky protocol that still need reverse engineering to be compatible with the new ACCESS radios

Here is the reply i've got from FrSky support on January 20th 2020 :

_"Thanks contact us, the D16 mode of the new ACCESS radio does not support third-party compatible receivers. You can use our XM, XM+ receivers
for the mini drone, then bind the receiver with the radio.

Best regards"_

My setup :
EMAX Tinyhawk 2 (Matekf411RX) :
- Betaflight v4.1.2
- FrSky_X-EU_LBT

FrSky X-Lite Pro :
ISRM v1.1.3 EU
open TX v2.3.5
protocol used => ACCST D16

Since FrSky doesn't seem to be willing to move, I hope that the Betaflight devs will soon find a solution to this issue :-)

Martin

taavituisk, 2020-01-22:
I have been flying fine with x-lite pro and betaflight 4.1.1 on tinyhawk s with the spi receiver. The has been some regression with Betaflight 4.1.2 and beyond where the link has become very unstable. The flight controller does not hang, i can see the rxloss message on the screen. The UK65 with the f3 performance edition works fine so far in D16 mode.

martimbrepost, 2020-01-22:
I downgraded from BF v4.1.2 to v4.1.1 and yes, the radio link is stable now even with telemetry on.
Does anybody know if the fix is going to be "re-applied" in the coming v4.1.4 ?
Or is it already fixed in the 4.1.3 that was just released ?

martimbrepost, 2020-01-22:
@mikeller 
Can we port de correction last effective in 4.1.1 into the next 4.1.4 ?
Sorry if this is a stupid question ;-)

mikeller, 2020-01-23:
@taavituisk, @martimbrepost: Can I get you guys to post a `diff` for the 4.1.2 configuration that is giving you problems? The only change in 4.1.2 that should be affecting SPI RX is #9218, and that one should not have negative side effects, so we'll need to figure out what is broken first.

martimbrepost, 2020-01-23:
@mikeller ,
here is my 4.1.2 diff with telemetry "on" that gives me the most frequent link losses : 

# version
# Betaflight / STM32F411 (S411) 4.1.2 Dec 22 2019 / 02:19:57 (89051e256) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF411RX   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF411RX
manufacturer_id MTKS
mcu_id 002900433139510c35343435
signature 

# name: BentProp

# feature
feature -AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# serial
serial 1 2048 115200 57600 0 115200

# aux
aux 0 0 0 1300 1700 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1700 2100 0 0
aux 4 28 1 1700 2100 0 0
aux 5 30 3 1700 2100 0 0
aux 6 39 4 1700 2100 0 0

# adjrange
adjrange 0 0 0 900 2100 29 0 0 0
adjrange 1 0 4 900 1600 12 4 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 100 200

# rxfail
rxfail 0 h
rxfail 1 h
rxfail 2 h
rxfail 3 h
rxfail 4 s 1000
rxfail 5 s 1000

# master
set gyro_sync_denom = 2
set acc_calibration = -61,0,37
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 2000
set rssi_offset = 20
set fpv_mix_degrees = 12
set rx_spi_protocol = FRSKY_X_LBT
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set failsafe_delay = 5
set bat_capacity = 450
set vbat_max_cell_voltage = 460
set vbat_min_cell_voltage = 320
set vbat_warning_cell_voltage = 330
set ibata_scale = 1050
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_cap_alarm = 360
set osd_alt_alarm = 50
set osd_vbat_pos = 321
set osd_rssi_pos = 6170
set osd_tim_2_pos = 2423
set osd_flymode_pos = 6177
set osd_throttle_pos = 2305
set osd_vtx_channel_pos = 6145
set osd_mah_drawn_pos = 2392
set osd_craft_name_pos = 6155
set osd_display_name_pos = 394
set osd_pid_roll_pos = 353
set osd_pid_pitch_pos = 34
set osd_pidrate_profile_pos = 186
set osd_warnings_pos = 14601
set osd_avg_cell_voltage_pos = 6497
set osd_disarmed_pos = 6346
set osd_flip_arrow_pos = 310
set osd_rate_profile_name_pos = 2113
set osd_pid_profile_name_pos = 2145
set osd_stat_max_spd = OFF
set osd_stat_max_curr = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_rssi_dbm = ON
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 1
set vtx_freq = 5732
set vcd_video_system = NTSC
set frsky_spi_tx_id = 54,175
set frsky_spi_offset = -56
set frsky_spi_bind_hop_data = 4,29,54,79,104,131,154,179,204,229,19,46,69,94,119,144,169,194,219,9,34,59,84,109,134,159,184,209,234,24,49,74,99,124,149,176,199,224,14,39,64,89,114,139,164,189,214,255,255,255
set frsky_x_rx_num = 4
set gyro_1_align_yaw = 1800
set name = BentProp

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 0
set dyn_lpf_dterm_max_hz = 250
set dterm_lowpass_hz = 100
set dterm_lowpass2_hz = 200
set vbat_pid_gain = ON
set feedforward_transition = 25
set acc_limit_yaw = 100
set crash_recovery = ON
set iterm_relax = RPY
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set iterm_windup = 40
set iterm_limit = 150
set p_pitch = 105
set i_pitch = 125
set d_pitch = 95
set f_pitch = 100
set p_roll = 105
set i_roll = 125
set d_roll = 95
set f_roll = 100
set p_yaw = 95
set i_yaw = 95
set d_yaw = 45
set angle_level_strength = 110
set horizon_transition = 80
set level_limit = 50
set horizon_tilt_effect = 80
set horizon_tilt_expert_mode = ON
set d_min_roll = 0
set d_min_pitch = 0
set auto_profile_cell_count = 1
set launch_control_mode = PITCHONLY

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 0
set dyn_lpf_dterm_max_hz = 250
set dterm_lowpass_hz = 100
set dterm_lowpass2_hz = 200
set feedforward_transition = 25
set acc_limit_yaw = 100
set iterm_rotation = ON
set iterm_relax = RPY
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set iterm_windup = 40
set iterm_limit = 150
set throttle_boost = 4
set p_pitch = 70
set d_pitch = 55
set f_pitch = 70
set p_roll = 70
set i_roll = 90
set d_roll = 55
set f_roll = 70
set p_yaw = 75
set d_yaw = 13
set f_yaw = 80
set angle_level_strength = 60
set horizon_level_strength = 30
set horizon_transition = 80
set level_limit = 30
set d_min_roll = 0
set d_min_pitch = 0
set auto_profile_cell_count = 2

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_expo = 50
set roll_rc_rate = 70
set pitch_rc_rate = 70
set yaw_rc_rate = 90
set roll_expo = 20
set pitch_expo = 20
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60
set tpa_rate = 25
set tpa_breakpoint = 1650

rateprofile 1

# rateprofile 1
set thr_mid = 20
set thr_expo = 50
set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set roll_expo = 60
set pitch_expo = 60
set yaw_expo = 20
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60
set tpa_rate = 45
set tpa_breakpoint = 1450
set throttle_limit_type = SCALE
set throttle_limit_percent = 75

rateprofile 2

# rateprofile 2
set thr_mid = 0

wingsuit, 2020-01-23:
@darkcody

Thanks! This is what it took to bind the BetaFPV Meteor 65 to my X-Lite Pro both with the latest firmware.

martimbrepost, 2020-01-23:
> @darkcody
> 
> Thanks! This is what it took to bind the BetaFPV Meteor 65 to my X-Lite Pro both with the latest firmware.

What are the latest versions that you applied ?
do you mean BF 4.1.3, OpenTX 2.3.5 and ISRM 1.1.3 ?

Do you have Telemetry activated ?
How many LiPos did you fly to confirm that it is all good ?

Thank you :-)

wingsuit, 2020-01-23:
BF 4.1.3, OpenTX 2.3.0 (Which is seemingly the latest firmware for the X-Lite Pro as shown on their website, or am I missing something?) and ISRM 1.1.3.

Telemetry is activated, radio is in FCC mode and I flew 5 batteries with no issues.

I am completely new to flying indoors however, and relatively new to quads in general so I may be unaware of how well the radio link should be performing. The quad flew flawlessly however.

martimbrepost, 2020-01-23:
> BF 4.1.3, OpenTX 2.3.0 (Which is seemingly the latest firmware for the X-Lite Pro as shown on their website, or am I missing something?) and ISRM 1.1.3.
>
> Telemetry is activated, radio is in FCC mode [...]

Thanks for the feedack.

OpenTX 2.3.0 on the Flysky website is the release that supports the new X-Lite Pro.
Since then, the OpenTX open source project released a new firmware (2.3.5).
It can be installed onto your radio using OpenTX Compagnon. But it is not mandatory ;-)

So weird that my Tinyhawk 2 and the one from one of my friends do not get a stable radio link with 4.1.2
- The BetaFPV Meteor65 uses the same Target firmware "MatekF411RX"
- You have the same radio as mine
- ...

so confused ?!?!?!?
Is there a hardware difference between the two boards ?

dariusintech, 2020-01-23:
Hey guys!

I tried 2 days in a row to bind HX-100SE with Taranis QX7. 
I followed the BetaFPV guide, but without succes. This -> https://www.youtube.com/watch?v=IcGbIt16L8Y
I reached with success the step where the binding green light is still, but nothing changes after that. 
The receiver is EU-LBT, confirm this info: 
http://support.betafpv.com/support/solutions/articles/27000051305-how-to-know-which-firmware-in-your-frsky-radio-controller-fcc-lbt-
In betflight configuration the default was FRSKY_X_LBT so I presume that the receiver is compatible with the Taranis.
Only different thing is that my Taranis doesn't have " ACCST D16". It's having only "D16".
I spend a lot of hours trying to find a solution to use this little quad, I joust bought a 5'' but I flight it 4 times because of the water.
Any ideas? Thank you!

taavituisk, 2020-01-23:
So, bumped betaflight to 4.1.3 on my tinyhawk s. No dropouts so far. Need to put more batteries through it to be sure.

taavituisk, 2020-01-24:
The rxloss drops are still there with the 4.1.3, i wonder how to debug it properly.

wingsuit, 2020-01-24:
@taavituisk how bad is it? I flew around 20 batteries outside today and only experienced one failsafe event when the drone was around 70 meters away up a slight incline close to the ground, around the same time that my video was cutting out. I didn't think much of it since I'm very new to micro drones and figured this is the limit of their capabilities. 

On a regular quad with a dedicated receiver I would expect much more, does cutting out at 70 meters indicate that I too am suffering from this bug or does it cut out much closer for you?

martimbrepost, 2020-01-25:
@wingsuit 
Distance is not a factor (in my case at least).
It happens randomly. Sometimes I can fly 2 or 3 lipos without drops.
look at the video, i'm just a couple feet from myself and RSSI is always above 80%.
[https://youtu.be/biuob6XKFBE](url)

Emax Tinyhawk II, telemetry off, BF 4.1.2, XLite pro ISRM 1.1.3 EU and OpenTX 2.3.5

mikeller, 2020-01-26:
@taavituisk, @martimbrepost: Can I get you guys to test with this firmware:

[betaflight_4.1.4_STM32F411.zip](https://github.com/betaflight/betaflight/files/4113348/betaflight_4.1.4_STM32F411.zip)

Unzip, then in the Firmware Flasher tab in configurator select 'MATEKF411RX(MTKS)' in the drop-down, 'Load Firmware[Local]', select the unzipped file, then 'Flash Firmware'.

Let me know if this fixes the problems you've been having since 4.1.2.

martimbrepost, 2020-01-26:
> @taavituisk, @martimbrepost: Can I get you guys to test with this firmware:
>
> [betaflight_4.1.4_STM32F411.zip](https://github.com/betaflight/betaflight/files/4113348/betaflight_4.1.4_STM32F411.zip) [...]

Great ! I will test it tonight when I get home.

martimbrepost, 2020-01-26:
@mikeller 
I whish I had better news. I had such high hopes.

I did flash the beta firmware (twice to be sure).
But the flight controller doesn't boot after that.
I had to foce it into DFU mode with the hardware BOOT button to recover and flash back v4.1.1

@taavituisk , did you get it to work ?

Did I do something wrong ?
- Tried with and without "Full chip erase"
- Selected MatekF411RX(MTKS)
- Load firmware from local and selected the unzipped file .hex
=> unknow usb device. Not even an LED on the FC ...

mikeller, 2020-01-27:
@martimbrepost, @taavituisk: So sorry for this, but there was a bug in the firmware that broke all F411 based boards: #9418

To my shame I have to admit that I did not test the firmware before I posted it. My apologies.

This one is tested now and does not brick the flight controller:

[betaflight_4.1.4_STM32F411.zip](https://github.com/betaflight/betaflight/files/4116103/betaflight_4.1.4_STM32F411.zip)

martimbrepost, 2020-01-27:
@mikeller 
Just amazing ! ALL FIXED :-)
Just flew 5 LiPos (about 25 min fly time. Thanks 48 khz)
Not even one little glitch.
It aslo looks like the RSSI signal is more stable and doesn't bounce between 85 and 99% anymore.

Michael, what did you do ? 
Is it something that was in fact broken from 4.1.1 ? Is it specific to ACCESS Radios ?
Can we expect to get this correction in the next 4.1.4 release ?

Thank you so much from myself and all my Tinyhawk friends 👍 .

mikeller, 2020-01-28:
@martimbrepost:

Glad to hear it all works now!

I am a bit surprised (in a positive way), because the version I posted above is just what is currently slated for 4.1.4, with the fix in #9418 

But if this is now all working fine then it looks like the changes in #9218, which went into 4.1.2 was what was required to address this problem.

JDredd81, 2020-01-28:
I have also tested with the beta 4.1.4 on a Tinyhawk 2 with a X9 Lite and the RXloss is still happening. Its better then 4.1.3 by far but still an issue. Please let me know what else I can test. When I go back to any 4.0 the problems are far less.

mikeller, 2020-01-28:
@JDredd81: Looks like the problems are somewhat hardware dependent. Can you please try with this one:

[betaflight_4.1.4_STM32F411.zip](https://github.com/betaflight/betaflight/files/4124804/betaflight_4.1.4_STM32F411.zip)
(latest in `4.1-maintenance` minus e28cf4323bf4eb9dd9cbdea04d40a09c0cce0478)

JDredd81, 2020-01-28:
Thanks. We have tried this and on X9 Lite t still get RX Loss.... I also tried on my older X9D+ (no access) and it will last longer but also drop after a couple of minutes...??? Almost like the it build up to much cache and then started to fail?

mikeller, 2020-01-29:
@JDredd81: Sorry, I overlooked the details in your last post. The ISRM TX module in the X9 lite is not yet fully supported - I am surprised that you even managed to get it to bind.
So can you please test with the X9D+ (with the internal XJR module preferably).
I think there was also a bug in some recent versions of OpenTX that led to connection problems after some time, so probably best if you make sure that your TX is up to the latest OpenTX version (2.3.5 at the time of writing). (However, do NOT upgrade the firmware on the XJT module to ACCST 2.0 - this is not yet supported by the FrSky SPI RX code.)

Asizon, 2020-01-29:
Done, I have tested this issue with latest OpenTx 2.3.5 and latest 4.1 maintenance. It seems to work well, I have only been able to observe two strange behaviors:
1.- When quadcopter is armed, all ok,  but if the quad is less than 1 meter from you, it is disarmed with a voice "telemetry lost".
2.- Spi Offset value is diferent than the previous version.
```set frsky_spi_offset = 54
set frsky_spi_offset = 64```

rotorgeeks, 2020-01-29:
For people having trouble binding with the newer FrSKY radios, do this
`set frsky_x_rx_num
bind_rx_spi`
After binding, it may be necessary to 'save' to keep the bind (IIRC with XJT Lite in D8 mode)

turboed13b, 2020-01-29:
You guys are better off going back to a tx with accst. I tried running an xlite pro with the xjt module and had a ton of issues and went back to the regular x lite.

The xlite pro binded in d16 but range was super short and you could feel the latency and have twitching. I switched to the xjt module to use d8 mode and it still couldnt be binded normally you had to go through the cli. I went to a Tiny Whoop race and it started suffering from low range issues too. The final straw was a couple new builds would failsafe within 20ft of the radio. 20ft in Tiny whoop world is pretty much half of a track.

JDredd81, 2020-02-04:
> @JDredd81: Sorry, I overlooked the details in your last post. The ISRM TX module in the X9 lite is not yet fully supported - I am surprised that you even managed to get it to bind.
> So can you please [...]

When will we see the compatibilities for the ISRM TX Module? I will test the X9D+ with the latest OpenTX to see if this problem changes. Thank you.

mikeller, 2020-02-04:
@JDredd81: I have got test hardware now and I am working on a fix - it's mostly working but there still are some glitches to be worked out.

JDredd81, 2020-02-04:
@mikeller nice to hear. Please let me know if you need any testers. Thank you.

kenthinson, 2020-02-16:
@mikeller I have been having issues with the CRAZYBEEF4FR and X-Lite connection. Just found this through google. Then checked my firmware version. It was the bugged version. Updated now through Betaflight . Thanks for fixing this :) 

I have not tested it yet but I can report a interesting find. The bugged firmware was using a high CPU amount. With 4.1.4 The CPU has dropped down to 20%. Much less than before. 

If I find anything else interesting after test flight I will report it as well.

Thanks again.

> @martimbrepost, @taavituisk: So sorry for this, but there was a bug in the firmware that broke all F411 based boards: #9418
>
> To my shame I have to admit that I did not test the firmware before I [...]

erNail, 2020-03-22:
I'm also experiencing short range issues when using a CRAZYBEEF4FR (Mobula6) with Betaflight 4.1.5 and a Taranis X9D Plus 2019 in ACCST D16 Mode. I failsafe inside when flying into the next room, and outside after flying around 20 meters away from me.

[54 more replies]

closed 2026-09-26 by blckmn