betaflight/betaflight #8605: JHEMCU F7 - target JHEF7DUAL softserial doesn't work.

yo8aiv, 2019-07-24:
hello.
hi. seems i **can't enable softserial port/s** for some reason. is there something i'm missing ?? 
just bought this board but without the current sensor (see diagram).

-seems in this schematic there are some errors on VTX wiring .. some users might make that mistake.
instructions seem limited on BG

camera control works on BF 4.0.4 with caddx F2 (after resource allocation to M6).
-would also like to know how i can use smartport on it with an XSR (so would like to reassign TX2 to use for telemetry).  No point in taking up a uart just for this , since tx2 is already free.

-cpu goes 70degC  is there a way to underclock.!? (no priority)

Thanks in advance.  
Keep up the good work. 

Update: got some updates on wiki page thanks. Created for the maintainer.

[image]

mikeller, 2019-07-24:
@atgfpeyv: Can you have a look into this please?

etracer65, 2019-07-24:
The target was created without softserial support.

No, there's no way to "under clock".

From a Betaflight perspective there's no reason why you can't use TX2 for smartport telemetry. I don't know if the board physically exposes UART2 though. 

The rest of your questions should really be directed at the board manufacturer.

yo8aiv, 2019-07-25:
thanks for your answers. 
srry forgot to update  >> after meesing  with softerial sport didnt work. i reflased and all good.
 **Update:** got s.port working on a uart using TX2 pad.

atgfpeyv, 2019-07-25:
The current sensor has a port current (CUR) pad connection.
The signal input configuration of the receiver is above UART2 (R2C). The telemetry signal can use other UART ports, but the TX2 port is not recommended.
The CPU temperature of 70 degrees refers to the core temperature of the MCU. Of course, if the BEC on the flight controller is large, the temperature of the entire board will rise.

LeoBenoist, 2019-07-25:
@yo8aiv can you provide you betaflight config for your UARTs (to see how you configured the smartport) (and port remapping if you have done some). Have you solderer smarport on T2 ?

yo8aiv, 2019-07-25:
here a diff all : 
# diff all

# version
# Betaflight / JHEF7DUAL (JH7D) 4.0.4 Jun 30 2019 / 14:26:13 (f3a95efa3) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name JHEF7DUAL
manufacturer_id 
mcu_id 002b004c3237510731313935
signature 

# name
name yo8aivF7

# resources
resource MOTOR 5 NONE
resource CAMERA_CONTROL 1 C09

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RSSI_ADC
feature GPS

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 32 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# led
led 0 5,11::AO:0
led 1 5,12::AO:0
led 2 8,12::AO:0
led 3 8,11::AO:0

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 5 900 1175 0 0
aux 2 2 5 1300 1700 0 0
aux 3 46 6 1700 2100 0 0
aux 4 13 4 1525 2050 0 0
aux 5 15 4 1125 1825 0 0
aux 6 19 7 1700 2100 0 0
aux 7 26 0 1800 2100 0 0
aux 8 40 1 1700 2100 0 0
aux 9 41 1 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 10 s 2000
rxfail 15 s 800

# display_name

# master
set gyro_to_use = SECOND
set acc_calibration = 28,-68,22
set rssi_channel = 16
set fpv_mix_degrees = 25
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = GPS-RESCUE
set ibata_scale = 300
set ibata_offset = -3
set yaw_motors_reversed = ON
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_rescue_angle = 35
set gps_rescue_allow_arming_without_fix = ON
set deadband = 5
set yaw_deadband = 5
set ledstrip_visual_beeper = ON
set osd_cap_alarm = 1500
set osd_vbat_pos = 6465
set osd_rssi_pos = 14424
set osd_link_quality_pos = 122
set osd_tim_1_pos = 2423
set osd_flymode_pos = 6509
set osd_g_force_pos = 2393
set osd_throttle_pos = 6338
set osd_vtx_channel_pos = 2081
set osd_ah_pos = 2126
set osd_current_pos = 6497
set osd_gps_speed_pos = 6393
set osd_gps_lon_pos = 6160
set osd_gps_lat_pos = 6144
set osd_gps_sats_pos = 2105
set osd_home_dir_pos = 14414
set osd_home_dist_pos = 10286
set osd_altitude_pos = 6359
set osd_avg_cell_voltage_pos = 10636
set osd_rtc_date_time_pos = 101
set osd_flip_arrow_pos = 4323
set osd_core_temp_pos = 2361
set osd_stick_overlay_left_pos = 8358
set osd_stick_overlay_right_pos = 8368
set osd_stat_rtc_date_time = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_flight_dist = ON
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_pit_mode_freq = 5740
set vcd_video_system = NTSC

# profile
profile 0

set vbat_pid_gain = ON
set p_pitch = 58
set d_pitch = 35
set p_roll = 40
set i_roll = 40
set d_roll = 30
set p_yaw = 70
set d_yaw = 20

# profile
profile 1

set vbat_pid_gain = ON

# profile
profile 2

# restore original profile selection
profile 1

# rateprofile
rateprofile 0

set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60

# rateprofile
rateprofile 1

set roll_srate = 63
set pitch_srate = 63
set yaw_srate = 63

# rateprofile
rateprofile 2

set roll_srate = 66
set pitch_srate = 66
set yaw_srate = 66

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75

# rateprofile
rateprofile 5

set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# restore original rateprofile selection
rateprofile 2

# save configuration
save
#

yo8aiv, 2019-07-25:
> The current sensor has a port current (CUR) pad connection.
> The signal input configuration of the receiver is above UART2 (R2C). The telemetry signal can use other UART ports, but the TX2 port is [...]

thanks for your fast answer. 
temperature related i don't imagine it will be a problem since quad will be flying but still that alarm worryies me when quad is static ist goes on in like 4min. 
- why ? > TX2 port is not recommended.  i will try reconfigure then TX1 to s.port and R2C (rx2) to sbus.

yo8aiv, 2019-07-27:
as advised swapped and works (no maiden yet): 
sbus - r2c  (rc controls ok)
s.port - tx1 (telemetry ok)

LeoBenoist, 2019-07-27:
@yo8aiv Did you check you motor order ? Mine where all in the wrong order. (This board is replacing an Hollybro board with a 4 in 1 esc with motor soldered in the correct order) My resource config:

```
resource MOTOR 1 B04
resource MOTOR 2 B00
resource MOTOR 3 B03
resource MOTOR 4 B01
```

yo8aiv, 2019-07-27:
> @yo8aiv Did you check you motor order ? Mine where all in the wrong order. (This board is replacing an Hollybro board with a 4 in 1 esc with motor soldered in the correct order) My resource config:
> 
> ```
> resource MOTOR 1 B04
> resource MOTOR 2 B00
> resource MOTOR 3 B03
> resource MOTOR 4 B01
> ```

yes had the same issue. just swapped wires.

LeoBenoist, 2019-07-27:
I don't have leds, the temps is like you said quite hot about 70C.

yo8aiv, 2019-07-27:
@LeoBenoist .. i initially didn't use the softmounts since i have a tight build . did you have any issue with stuttering motors ? .. yaw noise seems to make a mess.. weird cos motors and props are new.

LeoBenoist, 2019-07-27:
I haven't tryed it yet with prop! Keep you updated as soon as I can get out of the city!

yo8aiv, 2019-07-30:
> I haven't tryed it yet with prop! Keep you updated as soon as I can get out of the city!

ok. added the softmounts and flyies better no more stuttering.  
so nice to have  improvements would be  softserial at least 2 bidirectional wires for vtx and cam  and one additional (to relocate a tx from a uart where we use already rx) for s.port would be great.

LeoBenoist, 2019-07-30:
@atgfpeyv your answer on that?

yo8aiv, 2019-08-09:
@atgfpeyv could i ask if possible you add support in iNav please. 
i will try to make a request on github.
thanks.

LeoBenoist, 2019-08-11:
@yo8aiv No vibration problems for me. In the end I am quite please. I am using it to replace a bunt Kakute.

mikeller, 2019-10-01:
@yo8aiv: Has this issue been addressed? If so, can you close this please?

yo8aiv, 2019-10-12:
hi sorry for the late answer. i'll give it a try and report back. 
THX.

atgfpeyv, 2019-10-27:
@yo8aiv Working well now?

flybabo, 2020-01-26:
@atgfpeyv 
I couldn't find FC_CAM (PB08) pad on the board.  Where is it?
Four round shape labeled soldering pads (K,D,+,-) must be the debugger interface.
What are two round shape soldering pads near two micro JST connectors?

psyc0, 2020-05-03:
> hello.
> hi. seems i **can't enable softserial port/s** for some reason. is there something i'm missing ??
> just bought this board **JHEMCU F7 Deluxe** (see diagram) not the one with the current sensor .
> -seems in this schematic there are some errors on VTX wiring .. some users might make that mistake.
> instructions seem limited on BG
> 
> camera control works on BF 4.0.4 with caddx F2 (after resource allocation to M6).
> -would also like to know how i can use smartport on it with an XSR (so would like to reassign TX2 to use for telemetry). No point in taking up a uart just for this , since tx2 is already free.
> 
> -cpu goes 70degC is there a way to underclock.!? (no priority) _with BF 4 alarm on osd if quad stationary_
> 
> Thanks in advance.
> Keep up the good work.
> 
> **Update:** _got some updates on wiki page thanks @mikeller . Created for the maintainer._
> **Update:** got s.port working on a uart using TX2 pad.
> 
> [image]

@yo8aiv which pad was the camera control changed to, I guess B08 is not connected but its difficult to see, did you need to add a resistor and cap when you reassigned the pin?

ashgladiator, 2022-08-30:
Hello everyone.. I have the same FC. My issue with the telemetry RxBt not shown with the radio sensor, my set up is tx16s and frsky r9 module with r9 receiver and I loaded with ExpresLRS software.. please if you could help me