betaflight/betaflight #8608: Seamless revirsibility motors mode in angle mode

NietSem, 2019-07-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Please note that feature requests are not 'fire and forget'. It is a lot more likely that the feature you would like to have will be implemented if you keep watching your feature request, and provide more details to developers looking into implementing your feature, and help them with testing.

NietSem, 2019-07-25:
If a drone (with prop guards) crashes into a wall, it flips against the wall and slides down to the ground. What if u could reverse the back or top motors, so it could go back to level, even in situations like these. 
One drone on the market has implemented that on their drone. Video: https://youtu.be/hW1Fn32JBls?t=77

etracer65, 2019-07-25:
Normal brushless motors and ESCs that you would buy for a typical quad don't have the ability to reverse smoothly like that. It takes special (and expensive) instrumented motors.

NietSem, 2019-07-25:
Oke so it could then not be seamless? but like in 3d mode, u can also change the direction. So why wouldnt this be possible then?

etracer65, 2019-07-25:
Not really sure what you're looking for. Self-level will already bring the quad back to level - even if it's upside-down in flight. Crash-flip mode already exists to right yourself if you've crashed and are upside-down.

NietSem, 2019-07-25:
If a drone bumps in a wall to hard, the drone will get vertical. The drone would not be able to correct itself, because all the motors pushes the drone against the wall. Then the drone slides allong the wall down to the ground. But if you could reverse the motors, u could get off the wall.

etracer65, 2019-07-25:
Well no, that's not how it works. If you were to recenter the sticks the back motors will reduce in speed and the fronts will increase to level out the quad. Since the back motors are producing less thrust gravity will cause the back of the quad to lower. Reverse thrust is not needed.

jflyper, 2019-07-25:
I think I know what OP is saying.

If a craft gets completely vertical with craft's up direction against the wall, props would produce forces against the wall no matter how they are spinning.

OP wants to detect this situation and reverse the upper motor(s) so that it escape from the vertical attitude.

closed 2019-08-31 by stale[bot]