betaflight/betaflight #8611: BETA FLIGHT 4 not working with JETI EXBUS

Trexdominator700l, 2019-07-26:
BETA FLIGHT 4 not working with JETI EXBUS.

Version 4 doesn't seem to work with JETI exbus, but the same hardware and setup does work with 3.5.

When on the receiver tab nothing happens in 4 but in 3.5 it all works perfectly.

etracer65, 2019-07-26:
Please provide the information requested when you opened the issue.

> **Describe the bug**
> A clear and concise description of what the bug is.
>
> **To Reproduce**
> Steps to reproduce the behavior:
>
> **Expected behavior**
> A clear and concise description of what you expected to happen.
> 
> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
> 
> **Setup / Versions**
>  - Flight controller [what type is it, where was it bought from];
>  - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
>  - how are the different components wired up.
> 
> **Additional context**
> Add any other context about the problem here.

Trexdominator700l, 2019-07-26:
I'll re post once I have the required information

closed 2019-07-26 by Trexdominator700l