betaflight/betaflight #8612: Can't enable 32khz gyro after betaflight 4.0

beescuit, 2019-07-26:
**Describe the bug**
With versions before Betaflight 4.0, it is possible to enable the 32khz sampling mode on the KAKUTE F7 FC, but after 4.0, it just disables itself after restarting. 

**To Reproduce**
Flash any betaflight firmware greater than or equals to 4.0 onto a KAKUTE F7 and try to enable 32khz sampling mode

**Expected behavior**
After restarting, the 32khz gyro sampling mode should be enabled as it is in versions before 4.0

**Flight controller configuration**
```
# diff

# version
# Betaflight / KAKUTEF7 (KTF7) 4.0.4 Jun 30 2019 / 14:27:40 (f3a95efa3) MSP API: 1.41

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end
```

**Setup / Versions**
 - Flight controller: KAKUTE F7 AIO

RipperDrone, 2019-07-26:
No bug, has been dropped - limited value visa dynamic and ESC telemetry rpm based filtering. Was just marketing hype by some manufacturers, never proven to have real benefit.

beescuit, 2019-07-26:
oh, thanks, closing then

closed 2019-07-26 by beescuit