betaflight/betaflight #8617: OMNIBUS missing spektrum_sat_bind after flash

henrytseng, 2019-07-27:
**Describe the bug**
Looks like the CLI command for spektrum_sat_bind is missing after I reflashed with OMNIBUS.  

**To Reproduce**
Connect USB, flash with OMNIBUS ver 4.0.4.  Run CLI mode and try setting `set spektrum_sat_bind = 9` but receive ```###ERROR: INVALID NAME###```

**Expected behavior**
Previously had OMNIBUS 3.2.x loaded on a Eachine Aurora 90.  It runs a STM32F3.  And used the `spektrum_sat_bind` to set the receiver in binding mode.  

**Flight controller configuration**
Here is my previous confirutation:
```
{
	"generatedBy": "10.5.1",
	"apiVersion": "1.36.0",
	"profiles": [
		{
			"PID": {
				"controller": 1
			},
			"PIDs": [
				[
					40,
					40,
					30
				],
				[
					58,
					50,
					35
				],
				[
					70,
					45,
					20
				],
				[
					50,
					0,
					0
				],
				[
					15,
					0,
					0
				],
				[
					34,
					14,
					53
				],
				[
					25,
					33,
					83
				],
				[
					50,
					50,
					75
				],
				[
					40,
					0,
					0
				],
				[
					55,
					55,
					75
[... 2424 lines cut ...]
		"dshotBeaconTone": 0,
		"dshotBeaconConditions": {
			"_beepers": [
				{
					"bit": 1,
					"name": "RX_LOST",
					"visible": true
				},
				{
					"bit": 9,
					"name": "RX_SET",
					"visible": true
				}
			],
			"_beeperMask": 0
		}
	}
}```

**Setup / Versions**
 - Flight controller STMF3
 - Other components Transmitter is using DSMX via Specktrum2048.  

**Additional context**
All online posts say to use OMNIBUS firmware.

etracer65, 2019-07-27:
The feature was removed from many F3 builds because they are out of space. Recommend upgrading to a F4 or F7 flight controller. Or optionally use and older version of Betaflight to bind.

henrytseng, 2019-07-27:
Ok.  I'll restore to older firmware.

henrytseng, 2019-07-27:
what's the build TARGET for OMNIBUS STM32 F3?

etracer65, 2019-07-27:
`OMNIBUS`

closed 2019-07-27 by mikeller