betaflight/betaflight #8620: Unified Target always shows SD card

Docteh, 2019-07-28:
**Describe the bug**
SD card is showing up in status and in the blackbox tab, it shouldn't be.

**To Reproduce**
* Flash STM32F405 I tried with jenkins 1762.
~~* apply unified config that doesn't have an sd card~~ unnecessary
* `status` cli command and blackbox tab on configurator show a failed SD card

**Expected behavior**
Since `sdcard_mode = OFF` is the default, I'd expect the SD card to not show up in `status`

Also applies to bare STM32F405 target. Tried jenkins 1762

**Flight controller configuration**
None. `STM32F405` with full chip erase

**Setup / Versions**
Bare board.

**Additional context**
```
# status
MCU F40X Clock=168MHz (PLLP-HSI), Vref=3.30V, Core temp=41degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3732, Max available config: 16384

System Uptime: 12 seconds, Current Time: 2019-07-28T18:54:12.502+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 33, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Startup failed
Arming disable flags: NOGYRO RXLOSS CLI MSP
```

Possibly not related, but I was clicking around the configurator looking for a button to put the fc into MSC mode

mikeller, 2019-09-05:
@Docteh: I think this was fixed in #8656. Can this be closed?

Docteh, 2019-09-05:
Hi, not quite.

I can confirm that there is no mention of SD card in the configurator, ✨

I still see the line about the SD card in status. 

```
# status
MCU F40X Clock=168MHz (PLLP-HSI), Vref=3.30V, Core temp=50degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3781, Max available config: 16384

System Uptime: 11 seconds, Current Time: 2019-09-05T04:19:27.166+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 33, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: Startup failed
Arming disable flags: NOGYRO RXLOSS CLI MSP
```

on an unrelated note: in the configurator two tabs don't work with a bare target "Motors" and "Tethered Logging" 
```
msp.js:210 MSP data request timed-out: 104 MSP_MOTOR
msp.js:210 MSP data request timed-out: 116 MSP_BOXNAMES
msp.js:210 MSP data request timed-out: 150 MSP_STATUS_EX
msp.js:210 MSP data request timed-out: 110 MSP_ANALOG
```

Docteh, 2019-09-05:
In #8656 you're using `sdcardConfig()->mode`

Can you do something simular in `cliSdInfo`?
https://github.com/betaflight/betaflight/blob/2553a4a1f9a87270f39aec00702249cb425e7b00/src/main/cli/cli.c#L2316-L2330

MSP.send_message(104,undefined,undefined,undefined,undefined);

I think the MSP_MOTOR thing actually crashes the board, I'll make a separate issue for it.

PR #8849 (2019-09-09): Fixed 'sd_info' showing incorrect information when SD card not configured.
Fixes #8620.
closed 2019-09-09 by mikeller