betaflight/betaflight #8624: Failsafe 5 sec after GPS Rescue Triggers - expected behavior

callagga, 2019-07-29:
If you trigger GPS Rescue manually by a switch (failsafe mode) when the OSD is showing "Rescue N/A" should it then (which is did for me):  (a) go into rescue mode with all going well, heading home climbing, and then (b) about ~5 seconds later drop out of the sky?   It was however also just to pass 100m from home at the time too.  Note sanity checks were on.

Video: [https://youtu.be/kowWgEf2vRY]

Wondering:

1. Is this expected behavior? (or a bug)

2. Was the issue something to do with sanity checks being on?  e.g. after finding not enough satellites does it give a grace period?   Or was it something to do with almost being at the point of crossing the 100m from home (which I assume it shouldn't be).  

3. How could I have recovered: Would it be to quickly (a) arm then (b) turn off "failsafe" test switch?  [I have two switch Arm sequence with a delay for safety so this would chew up some time - as you see in the video not much time to react]

4. Perhaps for testing I should (a) lower number of satellite required to say 5, and (b) turn sanity checks off.   Then if everything works well then consider turning sanity checks on for when GPS Mode is used as my normal failsafe...

Was on Betaflight 4.0.4, on Speedybee F4 board.

etracer65, 2019-07-29:
The OSD warning `RESCUE N/A` means that GPS Rescue is not available because you don't have the minimum number of satellites or didn't establish a good home position fix when arming (because of not enough satellites). When activated GPS Rescue gives up to 5 seconds to recover enough satellites to work and if that doesn't happen then disarms. That's exactly what happened and it worked as it should. The warning is to tell you that GPS Rescue won't work.

closed 2019-07-29 by etracer65

callagga, 2019-07-29:
Thanks for clarifying