betaflight/betaflight #8630: Feature request: Current Limiter

Mode-1, 2019-07-30:
Hi Devs,

there are many scenarios when a quad should be limited in power output, for instance:
- BLHeli32 without current sensors - so no limiting at esc possible (Lumenier Micro Lux V3)
- new high current lipos (50+/100+) used in older quads which are not suitable for so much amps 
- full thrust high power flyaways when hitting a solid target
- overpowered quads (yeah they do exist for non-racers)
- energy saving freestyle flying without watching amps all the time while ascending
- protection for AIO-flight controllers
- most likely : my grandson flying like a lunatic completely ignoring limits of his quads :-)

I'm aware that there are some sorts of throttle limiting in betaflight, but they don't serve
this purpose. Most quads have a common current sensor for overall amps, so why not use this ?

I'm aiming at something like in music, called brickwall-limiter. A limiter scaling down outputs as soon as they are exceeded.

Scaling down all motor outputs with a factor would be a start for a test. Like motor 1,2,3,4 = flightcontroller_motor_output * limiting_factor. Limiting factor would be "1" for non-excessive amps and (programmed_amps/measured_amps) for the next pid-cycle. In my opinion worth a quick'n dirty attempt :-)

I got 1,5" / 3 " and 5" quads ready for testing. All of them have XT30/60-current sensors.
I would be happy to test a prototype current limiter some day.

Do you think this can be done, or is it complete nuts for some reason ?
Cheers

etracer65, 2019-07-30:
The current sensor data is filtered to make it usable and provide a more "averaged" view for the user. This means the values change slowly and lags relative to the actual current values seen by the battery.  Therefore the data is far too slow and delayed to be useful at the PID loop level.

The `motor_output_limit` is a more appropriate tool to limit maximum current.

Mode-1, 2019-07-30:
thanks etracer65, thanks for the BF-internal view,
maybe a slow and filtered response can be specially usefull. Since i don't know "C" therefore i don't know betaflight internals, but for instance a 1 Hz to 10 Hz response would be enough to give a smooth limiter, a short current-spike of 0.1 sec to 0.3 sec does not ruin any cables or motors. One could also limit as fast as the filtered current sensor output is able to and then fade the limiting factor back to "1" even slower (below 1Hz).  This would still prevent permanently high current flying. The limiter does not need to be too precise (within 5 amps), in most cases it would be limiting lets say 100 amps maybe down to 40 / 50 / 60 whatever the pilot sets. So filtering of the limiting factor would most likely be needed anyway, which is the same like feeding a limiter with an allready filtered input.
The question is, whats the amps-filter corner frequency in betaflight ?
A scaling of all motors at the same time still would let the pid do its job at stabilizing the quad, what do you think ? This filter would not adress race pilots anyway, it would just prevent long term overpowering a freestyle quad.

etracer65, 2019-07-30:
The default filter for the current sensor is a 1Hz PT1.

You also have to consider the alternate current source from ESC telemetry instead of onboard ADC. The ESC telemetry is even slower and will only update every 80ms.

Personally I don't think this will work well, but if somebody wants to give it a try then feel free.

Mode-1, 2019-07-30:
[image]

btw this is a log-view of a lightweight 250mm quad when ascending. amperage does correlate to the motor outputs and i dont see much or any lag in it. The sensor used in this quad is a diatone integrated XT30. The signal is then lead to an analog input A01 normally used for LED in the LUXF4OSD. The amperage signal could be used to limit total current for instance to the dashed line.
If lucky this could be done with low effort. Unluckily i can't do it, i program in different languages.

part of my diff concerning A01 as current input:
board_name LUXF4OSD
resource MOTOR 5 NONE
resource SONAR_TRIGGER 1 NONE
resource LED_STRIP 1 NONE
resource ADC_CURR 1 A01

etracer65, 2019-07-30:
The problem is you're looking at it from a macro view while the PID controller may be running at up to 8000 times per second. Even 10ms of delay is 80 pid loop cycles where we're calculating the motor mixing to respond to flight requirements. Scaling the motor outputs based on old data (in the PID loop context) is a bad idea and will result in the PID controller fighting itself.

I think you need to use the `motor_output_limit` setting as it should do everything listed in your use cases. Its entire purpose is to limit the motors to a certain percentage which will proportionally reduce the maximum current.

Mode-1, 2019-07-30:
hi etracer65 - well you're right, this is a very global view on very fast data.
But when i throttle down manually then it's based on old (HUD) data too.
What about modulating the motor_output_limit slowly (like i would throttle down), does this lead to a tasmanian (pid) devil ? If not then why not use this instrument. I found #7482 where it is discussed as a per profile setting. Didn't know that one before...

etracer65, 2019-07-30:
You just linked to the pull request that added `motor_output_limit` like I've been telling you to use.

Mode-1, 2019-07-30:
... unfortunatly i can't use it. Browsed BF-code for "amperage" and "motor_output_limit" and found where it's used but that code looks like alien language for me. Gave a few tries on compiling BF here on my Windows 7 machine, but all that stuff is too complicated for C-noobs, so i give up. Thanks for showing me what to use for this feature etracer65.
I will keep this issue open for some more time just in case a more capable person wants to make it real, but i'm out, cheers all.

mikeller, 2019-07-30:
@Mode-1: All you need to do is to set `motor_output_limit` to the desired maximum percentage of motor output in CLI, no coding required.
You can probably get a good idea of what percentage you want to use by doing a short punch-out, and then checking the 'maximum current' stat after the flight. Then take the percentage of that that you want to be the maximum current, and enter it for `motor_output_limit`.

Mode-1, 2019-07-31:
@mikeller: setting motor_output_limit to lets say "75" via CLI will deliver a static limit. That's not what the feature would be all about. I got quads used with 3S and 4S depending where i fly. No matter what i would set, it would never ever limit the quad to 25 amps for instance. The 4S could result in 25A at 50% throttle whereas 3S maybe at 75% throttle for 25A. So for a working current limiter nothing can help except a dynamic adaption of motor_output_limit
What i got in mind was (i write a very primitive basic-code for my convenience) :

if amperage >= amps_max then                                   // will limit to amps_max if (?)
    motor_output_limit = (amps_max / amperage)        // motor_output_limit is linear with current
else
   if motor_output_limit < 100 then           // given 0.01 sec limiter loop time
       motor_output_limit += 1                   // recover to 100 in 1 sec worst case
    end if
end if

Maybe this would "pump" - anyway - worth a try in my opinion

mikeller, 2019-07-31:
In this case you can just set up one profile for every battery type you fly, and set an appropriate `motor_output_limit` for it. If you use `auto_profile_cell_count` you can even have the right profile for every battery type automatically selected.

And yes, it **will** pump when using current limiting, this is the very nature of current limiting - you get this even with lab power supplies if you run them in current limiting mode.

Mode-1, 2019-07-31:
Well there are algorithms for instance in music business doing proper limiting without pumping (dA/dt preview etc.), but i doubt there's enough space (and request) for implementing such presently. Next stage would then be power limiting which is a different beast. I will now close the issue, but i'm sure in the future this topic will become more interesting as battery tech will advance and maybe at some point you will have 8S-200/400c batteries delivering kilowatts without a proper limiter. Same applies for tiny whoops driven with 4-6S where some cables can only be 0,5mm² for weights sake. But you will see em cables and motors glow at night, a funky side-effect :-) Thanks all.

closed 2019-07-31 by Mode-1

mikeller, 2019-07-31:
> Well there are algorithms for instance in music business doing proper limiting without pumping (dA/dt preview etc.),

Yes there are. But as @etracer already remarked, with the output loop normally run 100s of times faster than the measurement loop, and the hardware doing the measurements not capable of running at output loop speeds, this is hardly applicable here without re-engineering the hardware to support this.