betaflight/betaflight #8639: Remapping uart to I2C1 does not work

demvlad, 2019-08-01:
I have SP Racing F3 flight controller. My UART2 socket  broke down. I2С1 socket is like to UART. 
I don't use it. I want to remap UART2 RX,TX to I2C1 SCL and SDA pins. I tried to do it. But after that all sensors (gyro, accel ...) do not work. Is it possible to do remapping UART2 to I2С1?

mikeller, 2019-08-01:
@demvlad: Short answer: No.

The pins used for I2C1 can not be used for an UART.

demvlad, 2019-08-01:
Thank you for information!
Can i use free motor pins (MOTOR5-MOTOR8) for remapping an UART?

mikeller, 2019-08-01:
@demvlad: Not sure - have a look at https://github.com/betaflight/betaflight/wiki/reference/stm/stm32f3_pins_timers_dma.pdf to see what pins can be used for an UART, and https://github.com/betaflight/betaflight/tree/master/src/main/target/SPRACINGF3 to see what pins are used on your board.

demvlad, 2019-08-02:
All good! This variant remapping UART2 pins is working: TX to A02 (MOTOR7), RX to A03 (MOTOR8).
Thank you for help!

closed 2019-08-02 by demvlad