betaflight/betaflight #8654: Blackbox Log Viewer 3.3.1 MacOS : error opening log

JimB40, 2019-08-03:
[image]

Got this message every time I want to open log from my R349.
BF firmware is 3.5.1
Tried different logging rates 500, 1000, 2000.
Installed 3.2 version -> same error
Log file in enclosure.

[BTFL_BLACKBOX_LOG_R349HD_20190803_143416.bbl.zip](https://github.com/betaflight/blackbox-log-viewer/files/3463869/BTFL_BLACKBOX_LOG_R349HD_20190803_143416.bbl.zip)

JimB40, 2019-08-04:
Guys thank you for help. Upgrading do 3.5.7 and using mass storage mode did the trick. I didnn't mass storage doesn't work properly on 3.5.1 (diatone shipping with this firmware) and "save to file" is prone to errors and will be depreciated.

closed 2019-08-04 by JimB40

yoodog, 2019-09-01:
same problem with me on 4.0.5 now. Writing to  SD card on built in port.
I got a file for each flight 10-23mb - 3-6min flights. But not a single one can be open.
I run f745 board Omnibus v2 f7- 8/8 dshot600 and RPM Filtering + standard 4.x.x filtering 
- blackbox set to 2khz, SD card = card is ready, and well it records
- black box switch is AUX2, same as ARM switch, so records every time I arm
- set debug_mode = GYRO_SCALED 
- blackbox_p_ratio = 64 (I didn't touch it, guess it it is default, but mentioning it as it seems to be a key of some sort)

my diff all incl defaults and dump
[dump full - not all may be good for your controller.txt](https://github.com/betaflight/betaflight/files/3563918/dump.full.-.not.all.may.be.good.for.your.controller.txt)

[Diff all defaults.txt](https://github.com/betaflight/betaflight/files/3563916/Diff.all.defaults.txt)

log file 
[LOG00012.BFL.zip](https://github.com/betaflight/betaflight/files/3563917/LOG00012.BFL.zip)

can any one advice