betaflight/betaflight #8662: Feature request: selection for different vtx

nightflasher, 2019-08-04:
I'm just wondering if it would be possible to add a menu for selecting the used vtx so the OSD doesn't display 600mW for e.g. 200mW?! I think the community would enter the values themselves. So if this small database is existing anybody can select the wattage they need, not referring to the documentation anymore.

Just asking

etracer65, 2019-08-04:
In Betaflight 4.1 all vtx settings already are user controlled and MUST be entered using the `vtxtable` command. This includes the bands/channels and power levels. The user configures this based on local laws and the capabilities of their vtx. There are no more default settings for vtx control in Betaflight 4.1.

Thinstation, 2019-08-31:
I see some of this code in the 4.0 maintenance branch, is this being back ported?

etracer65, 2019-08-31:
No

mikeller, 2019-08-31:
@Thinstation: The code for the VTX table functionality existed before 4.0 was released, but it did not yet support all different VTX hardware types that we support, so it was disabled in 4.0.

closed 2019-10-07 by stale[bot]