betaflight/betaflight #8674: Stop recording the Blackbox log AFTER disarm

McGiverGim, 2019-08-07:
**Is your feature request related to a problem? Please describe.**
Sometimes a problem occurs in the craft, and the quad disarms but the Blackbox log has been stopped before the problem gets recorded. The disarm event does not appear never in the log. This has been discussed in the Betaflight Slack chat.

**Describe the solution you'd like**
We usually have the Blackbox log attached to the arm/disarm switch. When you disarm the quad, the Blackbox must be stopped after the disarm has been executed. Maybe adding a little timer to stop it after the user has selected to stop log/disarm the quad will be enough.

**Describe alternatives you've considered**
Configure the Blackbox log mode as ALWAYS, but this seems overkill.

**Additional context**
Nothing.

etracer65, 2019-08-07:
The disarm event not showing in the log is not related to the timing. It's related to the disarm not being triggered by the arming mode switch - that's what makes the event in the log. So logging for a period after something else causes a disarm (like a runaway takeoff) still won't capture the disarm event marker in the log. So extending the log past the actual disarm in this case would add confusion since you wouldn't be able to actually tell when the disarm occurred.

What we really need is more robust "event" markers in the log. I suggested something like this quite a while ago but just never had a chance to work on it. Would also take some unknown amount of effort to properly display them in the Blackbox Viewer.

https://github.com/betaflight/betaflight/issues/5141

McGiverGim, 2019-08-07:
So then, we need to send some events when the event occurs, not when the switch is changed. At least to fix it a little.

ijustwannafly, 2019-08-08:
Maybe I don't understand how to setup blackbox properly, but on my dysf4pro boards with blackbox tied to my arm switch, every time I disarm the blackbox log stops from that point on.

So say I arm and trigger blackbox to start logging, I have never been able to disarm, make a setting change, and then re-arm and have my blackbox log continuous.

mikeller, 2019-08-08:
@ijustwannafly: That's exactly how the default behaviour of blackbox is designed to work: It logs whenever logging is enabled **and the craft is armed**. If you want this to be different, change `blackbox_mode` to `always`.

ijustwannafly, 2019-08-10:
@mikeller I tested the blackbox always mode and it sticks to its name.  It records blackbox data even when connected via usb and the blackbox is set to a mode switch.  Is this the proper behavior?

It would be nice if there was an option to just have a single blackbox file (continuous) of a single flight regardless of how many times the quad is disarmed and the logs should only start/restart when the switch is used.  Is this not possible?

mikeller, 2019-08-10:
@ijustwannafly: Yes, you are right - the `ALWAYS` mode was added for use on flight controllers in a test rig without an RX connected, which means it does not check the mode switch.
I guess we could add another mode `SWITCH_ONLY` or so that checks the switch, but not the arming status.

alexrjohnston, 2022-02-27:
SWITCH_ONLY would be super helpful for gyroflow logging as well, did this get implemented?

closed 2026-09-26 by blckmn