betaflight/betaflight #8676: Mamba F7 Bi-dir not working

HrrrPippi, 2019-08-07:
Hi there.
Im trying to setup bi-directional Dshot to make rpm filters work on a new build, but get an error RPM filter in the osd.

I can get the motors to spin for ½ a second in the Gui as well, then the firmware locks up. FC is still responding, but wont spin the motors again.

The hardware i'm using is mamba f7 and typhoon 32 4i1

I flashed v6.4 on to the esc. So it should work with BF 4.0.5, right?

Thanks for all your help.

HrrrPippi, 2019-08-07:
Sorry, i made a mistake, it is the mamba f7 mini fc i'm using.

kc10kevin, 2019-08-07:
Have you gone to the Wiki page and followed the directions?   Downloaded the config file, pasted it into the CLI and saved?   It has been tested and works great if you use config file.