betaflight/betaflight #8690: sdcardSdio_poll loops indefinitely if card reads fail

jflyper, 2019-08-09:
This is the problematic section.
https://github.com/betaflight/betaflight/blob/b5981109b5151f93df0c8172c809967fbc476165/src/main/drivers/sdcard_sdio_baremetal.c#L417-L427

When `sdcard_receiveDataBlock` returns
- `SDCARD_RECEIVE_ERROR `, `goto doMore` (top of outer `switch`) is executed and it will eventually comes back to this section and repeat the `goto doMore` while the error condition persists.

- `sdcardSDCARD_RECEIVE_BLOCK_IN_PROGRESS`, timeout is checked. If it timeout is not reached, it will `break` (line 419), and will effectively restart to outer `switch`. If timeout is reached, then it will fall through the `SDCARD_RECEIVE_ERROR` case, which effectively repeats the outer `switch`.

In summary, the `sdcardSdio_poll` does not return if a block read fails.

`scared_receiveDataBlock` is called again

jflyper, 2019-08-09:
Additional context:

The issue develops when 1-line wired SDIO is activated with `use4BitWidth = true`.

mikeller, 2019-08-09:
> The issue develops when 1-line wired SDIO is activated with use4BitWidth = true

I would assume that it will develop as well when a receive error happens that isn't caused by misconfiguration?

jflyper, 2019-08-10:
Yes, I would assume so. The logic is bad anyways, we should fix it to quit after repeated errors and shutdown the subsystem.

hydra, 2019-08-15:
good spot @jflyper

hydra, 2019-08-16:
This also happens when DMA transfers do not complete, as noted here - https://github.com/betaflight/betaflight/pull/8697#issuecomment-521957359