betaflight/betaflight #8692: Possible bug in codification of band/channel at the MSP_SET_VTX_CONFIG

McGiverGim, 2019-08-09:
**Describe the bug**
- Betaflight, into the CLI lets configure the `vtx_band` and `vtx_channel` from `0-8`.
- Into MSP, the MSP_VTX_CONFIG returns the channel and band in the same manner, `0-8` 
- The problem is that the MSP_SET_VTX_CONFIG lets configure the channel and band codifying it into a uint16 in this way: (band - 1) * 8 + (channel -1) so it is restricted to `0-7` and interpreted like `1-8`. This is the decodification:
https://github.com/betaflight/betaflight/blob/b5981109b5151f93df0c8172c809967fbc476165/src/main/msp/msp.c#L2305-L2307
I think this was not the initial idea, because the constant to decide if this field contains a band/channel or a frequency is like this:
https://github.com/betaflight/betaflight/blob/5123e8bb38f99ed6958529acf98cfc1c66827e05/src/main/drivers/vtx_common.h#L44-L46

**To Reproduce**
Try to write and read the band and channel zero for example using MSP.

**Expected behavior**
The band and channel must have the same range. And the constant must represent the maximum value. 

**Flight controller configuration**
Not needed.

**Setup / Versions**
- Current development branch.

**Additional context**
The MSP command has been published into currently released Betaflight versions, and I think is used by the LUA script. Is this a bug? Or simply a bad implementation and is better to continue like this?

McGiverGim, 2019-08-09:
I have verified and the constant is OK, 63, that is 7*8 + 7. So the only problem is that we can't send zero as value.

etracer65, 2019-08-09:
Yeah, this was just a poor original design. There are actually 9 valid values (0-8) so it's not possible to encode in 3 bits.

A couple of choices:
1. we could leave the broken implementation as is and simply add the band and channel as new separate bytes in the MSP.
2. Fix the encoding to use 4 bits each for band and channel.

For option 1 we leave broken ugly stuff but don't break anything using the MSP (like Lua). Although the Lua implementation is **probably** wrong I'd have to assume based on the misunderstanding of the data.

For option 2 we break Lua (assuming it's already kind of working) and probably other apps like SpeedyBee but the structure is fixed for the future.

@mikeller Any thoughts?

etracer65, 2019-08-09:
Commented in Slack - adding here for reference:

> Anticipating that the only thing we can do is add the standalone values for band, channel and frequency to the MSP_SET_VTX_CONFIG and just deprecate the old incorrect "packed" value, I updated my branch with the new values added to the end of MSP_SET_VTX_CONFIG.

So I went with option 1 for now.

mikeller, 2019-08-10:
Yeah, I'd go for 1. - MSP is a public API, and we don't know what is out there using it.
In this case, since changing VTX settings isn't something that can be done easily through OSD in many cases, I think it is particularly important not to break existing solutions like lua scripts.

I think this problem comes about because, in the original implementation of VTX control, band and channel were 1-based, having no '0' value. That's when the MSP for it was added, but it was never used by us in configurator. Later on, '0' was added as a special case to indicate that the frequency was set directly, ignoring band / channel presets. Because we did not use the MSP at this point, we did not notice that this did not work for the API's definition.

closed 2019-09-16 by stale[bot]