betaflight/betaflight #8701: Core temp on OSD shows unwanted "C" character

0crap, 2019-08-10:
From an old jflight I installed Jenkins build # 1797.

All seems to look fine except that I see an weird C character if I turn on "Core temp" on my OSD.
This has never been the case.
See picture.

I'm on a MATEKF405 (MKF4) FC.
I tried uploading a different OSD font, but the weird C character stays.
[image]

etracer65, 2019-08-10:
It's by design. It represents "C"ore temp. To distinguish it from other temperature readings like "E"sc temp.

McGiverGim, 2019-08-11:
And you must update your fonts with the latest master of Configurator, it will appear a temperature symbol too.

0crap, 2019-08-11:
Well, thx, that explains it!

Would not have guessed that one. Think because the C is so far apart from the temperature.
To be honest, it was nice and clean to have the temp in a corner.
Now it's a bit ugly. But I guess somebody asked for this.
Can't make everybody happy.

Is it an option to at least remove one space between the C and the actual numbers?

hydra, 2019-08-15:
Yeah, I wondered what that 'C' was all about too, I guess we won't be the last @0crap.

Gozzz, 2019-08-15:
And I was wondering what the E was that showed up next to my temp... Good to know.

mikeller, 2019-08-15:
For release notes: Include a note about the need to update the OSD font.

0crap, 2019-08-15:
@mikeller I'm sure you understand, but besides the release notes this is more about the C or E character are not needed and not intuitive.
To line it up cleanly on the left-side of the OSD is not possible anymore, because of the three added characters (spaces included).

I think it's preferred (at least I do) to remove the added characters in front of the temperature.

mikeller, 2019-08-15:
@0crap: I think this is not a question of need or intuition - it is just a matter of personal preference. Some people like their numbers with added symbols so they know what they stand for, others like to have just the numbers, yet others would like to have some of this and some of that.
At the moment Betaflight is leaning towards the 'added symbol' approach, with lots of special symbols for units available in our fonts. In this case it also makes to add a specification of what the displayed temperature stands for - after all this is just as relevant as the units.
I don't think there is a good way to make this configurable that will work without using a lot of memory and adding a lot of complexity, making maintenance of the OSD code harder. Sorry.

0crap, 2019-08-15:
I understand and I'm totally for the added Celcius symbol/font, perfect.
Just the 'C' for Core and 'E' for ESC including the extra spaces are a bit too much.

But OK, let me know if you want to close this issue, if so, I'll close it.

mikeller, 2019-08-15:
@0crap: Looks like you are one of the 'in the middle' guys then. :wink: 

As I said, this is a matter of taste / preference, we could discuss this all day long, and if we changed it somebody else would come along and ask to have it changed back because they like it better the other way round, and would be just as right as you.

But please leave this open for now - it will help remind me to add a note to the release notes telling users that they have to update their OSD font.

mikeller, 2019-09-05:
Added note to the release notes: https://github.com/betaflight/betaflight/releases/tag/4.1.0-RC1

This can be closed now.

0crap, 2019-09-06:
Perfect!

closed 2019-09-06 by 0crap