betaflight/betaflight #8704: Diatone Mamba f722s FC - AUX 1 signal default at 1875

simoccc, 2019-08-11:
Hi,

I wanted to report this issue that, if not specifically of my FC ( I'm digging with Diatone ), it's kind of dangerous/important

The summary is this:

if the FC is powered on without the radio powered up and binded , AUX 1 signal is randomly between 1825 and 1975 .
If the radio is powered on before no problem.
I haven't checked if powering up the FC without the RX gives the same problem.
I'm worried because AUX1 is the channel I use for ARM , and knowing that this value is randomly so high ( and maybe who knows if it gets higher ) my quad could arm if this combination of events happens.

Tried to update, re-flash the FC , same behavior.

I don't know if there is a way through CLI to setup one AUX channel default PWM value, if not I don't understand why it's going so high even without the attached RX powered up .

etracer65, 2019-08-11:
You didn't provide the required `diff` of your settings.

I'm assuming aux1 is your arming channel. If so then this is normal. When there is no radio signal the arming channel value will get fixed just above or below the arming range to ensure it stays in the non-armed range.

simoccc, 2019-08-11:
> You didn't provide the required `diff` of your settings.
> 
> I'm assuming aux1 is your arming channel. If so then this is normal. When there is no radio signal the arming channel value will get fixed just above or below the arming range to ensure it stays in the non-armed range.

Sorry for the missing diff.
What you are telling is right, AUX 1 is my arming channel.
So in general this is a normal behavior , I missed it totally maybe its new in 4.0 BF this is my first 5" build with it...
Thank you and sorry for the false alarm. You can close it

closed 2019-08-11 by simoccc

etracer65, 2019-08-11:
This is not new behavior. Been this way forever.

simoccc, 2019-08-14:
> This is not new behavior. Been this way forever.

You are absolutely right, now I realize it even in the old BF build I have...sorry for the false alarm