betaflight/betaflight #8708: Support for DJI (style) GPS (one UART for both "GPS and Mag")

JulianGoesPro, 2019-08-11:
Hello :)

I am coming from iNav and noticed BetalFight does not support the Naza GPS protocol (which uses one UART for both "GPS and Mag") which I think is kind of sad (already using SoftSerial for vTx control on the Mamba Mini Stack)

Maybe its an easy pull copy paste job to move it from iNav to BetaFlight so I thought I would ask...

BTW, not only DJI is using it: https://www.banggood.com/Ublox-M8N-GPS-Compatible-with-DJI-NAZA-Lite-V1-V2-Flight-Controller-Phantom-1-2-Vision-p-991718.html

Thanks a lot :)

Cc2Gre, 2019-09-26:
Great !

seriousrob, 2020-03-27:
If you need anyone to test, i have an iNAV setup using the SpeedyBee F4 and the NAZA GPS. It works great except DSHOT protocol. That's why i'd love to go back to betaflight with it. Well, as soon as we are allowed to go out again.....

closed 2026-09-26 by blckmn