betaflight/betaflight #8710: Slight delay spinning up after arming

kevindehecker, 2019-08-11:
**Describe the bug**
I'm noticing a slight delay in spinning up the motors just after arming. I have done tests on a Trashcan, we are talking about 0.15s after a arming command arrives before the motors do a first movement.

To compare, if I put the same quad in 3d mode and measure how much time it takes to change direction of the motors, this is in the order of 0.03s. 

A video illustrating these two cases: https://youtu.be/UTKFt7oXics
In the first clip, the leds going off indicate the command to arm has been received.

**To Reproduce**
Steps to reproduce the behavior: turn off motor_stop, arm, measure delay between arming and motors spinning up.

**Expected behavior**
I expect the motors to spin up as fast as possible, which I expect to be similar or faster than changing directions in 3d mode.

**Flight controller configuration**
```
# diff

# version
# Betaflight / CRAZYBEEF4FR (C4FR) 4.0.0 Dec 11 2018 / 01:44:17 (3b479f92d) MSP API: 1.41

board_name CRAZYBEEF4FR
manufacturer_id 

# name
name Trashcan

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature -OSD
feature -ANTI_GRAVITY
feature -DYNAMIC_FILTER
feature LED_STRIP
feature AIRMODE

# beeper
beeper -ON_USB

# beacon

# map

# serial
serial 0 2048 115200 57600 0 115200

# led
led 0 4,7::CT:1
led 1 5,7::CT:1
led 2 6,7::CT:1
led 3 7,7::CT:1
led 4 8,7::CT:1

# color

# mode_color
mode_color 6 0 0
mode_color 6 1 2
mode_color 7 0 7

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 900 2100 0 0
aux 2 15 3 900 1425 0 0
aux 3 28 1 1550 2100 0 0
aux 4 35 2 1550 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_trim_roll = 42
set rc_interp_ch = RPY
set motor_pwm_protocol = DSHOT600
set vbat_min_cell_voltage = 29
set vbat_warning_cell_voltage = 30
set ibata_scale = 1175
set yaw_motors_reversed = ON
set 3d_deadband_low = 1450
set 3d_deadband_high = 1500
set 3d_neutral = 1425
set 3d_deadband_throttle = 3
set osd_warn_esc_fail = OFF
set osd_vbat_pos = 2102
set osd_rssi_pos = 2083
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2134
set osd_flymode_pos = 2456
set osd_throttle_pos = 2423
set osd_crosshairs_pos = 2253
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2432
set osd_craft_name_pos = 2443
set osd_warnings_pos = 329
set osd_disarmed_pos = 2346
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 0
set vtx_low_power_disarm = ON
set vtx_freq = 5917
set frsky_spi_tx_id = 1,0
set frsky_spi_offset = -41
set frsky_spi_bind_hop_data = 0,2,4,6,8,10,12,14,16,18,20,22,24,26,28,30,32,34,36,38,40,42,44,46,48,50,52,54,56,58,60,62,64,66,68,70,72,74,76,78,80,82,84,86,88,91,92,0,0,0
set frsky_x_rx_num = 1

# profile
profile 0

set throttle_boost = 0
set level_limit = 70

# rateprofile
rateprofile 0

set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 130
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 80
```

**Setup / Versions**
 - Flight controller: original Trascan with Crazybee F4  
https://www.banggood.com/Eachine-TRASHCAN-75mm-Crazybee-F4-PRO-OSD-2S-Whoop-FPV-Racing-Drone-Caddx-Eos2-Adjustable-Camera-25200mW-VTX-p-1386071.html?ID=530009&cur_warehouse=CN  
 - BLHeli version: S-H-50, 16.7

**Additional context**
The two clips in the video above were made with the same hardware, BF-, and BLHeli settings. Except of course for the 3d mode particulars. I played around with all BLHeli settings and BF I could find but did not see a significant change. I disabled rc-smoothing on the throttle channel to be sure.

In the BF code I could not find a good reason for this delay, so it could be the escs after all. (I earlier did some tests with a brushed whoop which seemed faster, but there were other considerations here). 
Considering that in 3d mode the motors can actually change direction amazingly fast, I was hoping someone could point we in the right direction on where to look...?

mikeller, 2019-08-11:
@kevindehecker: Have a look at https://github.com/betaflight/betaflight/blob/b310f9b34873b5ef422152d88eb6a789af991538/src/main/fc/core.c#L406. There is a 1.2 second delay before arming to make sure all Dshot beep commands have finished before the commands to set the spin direction for crash flip are sent. This is to avoid the situation of props spinning up in the wrong direction because the direction commands were missed by the ESC still being busy processing the beep command.

etracer65, 2019-08-12:
@mikeller That only applies if the beacon is active. In this case the `beacon` command output in the `diff` would indicate it wouldn't be active.

@kevindehecker What's the concern about 0.15 seconds (as you measured it)? There are various things that need to be configured with inherent delays when arming. Additionally the ESC firmware does some control signal validity checks before starting the motors.

The majority of that time is likely due to the DSHOT commands sent during arming to set the proper motor directions (normal or reversed depending on arming vs. turtle mode) and to enable/disable bidirectional telemetry. Sending DSHOT commands takes some time. In fact each DSHOT command is sent 10 times with a 10ms delay so in reality on BF4.0 the delay would be closer to 0.2 seconds. On BF4.1 the second command isn't needed so the delay will reduce by about 0.1 seconds.

kevindehecker, 2019-08-12:
Thanks, yeah, I saw that one. Didn't think it was it because I'm seeing a smaller delay of 0.15s. Tried it out anyway just now by setting `#define DSHOT_BEACON_GUARD_DELAY_US 1`, but it did not have any effect.

kevindehecker, 2019-08-12:
@etracer65 we are doing a kind of scramble and chasing game, in which motors are off to save energy but where reaction time is of upmost importance. Especially at take off, otherwise the target is out of sight.

Can you point me to this 10x sending of dshots commands in the code? Can't seem to find it...

BTW, I'm seeing the same spin up delay when motor_stop is enabled (and arming is already done minutes ago). Is this 10x sending also happening then?

I will try 4.1 now, see if that is faster.

kevindehecker, 2019-08-12:
Too bad, BF 4.1 (latest master) has exactly the the same delay of 0.15s.

kevindehecker, 2019-08-12:
Wait, I spoke a too soon. Actually the new delay of BF 4.1 seems to be consistently smaller, just not that much. I'm now down to 0.12 :)

(I was doing another test at the same time, increasing the pid loop to 8khz. This actually bumped me down to 0.17s, which confused me. )

etracer65, 2019-08-12:
@kevindehecker 

> BTW, I'm seeing the same spin up delay when motor_stop is enabled (and arming is already done minutes ago). Is this 10x sending also happening then?

No. Any delay here is purely in the ESC. As I mentioned, the ESC does signal validation checks before starting the motors and we have no control over that.

> Wait, I spoke a too soon. Actually the new delay of BF 4.1 seems to be consistently smaller, just not that much. I'm now down to 0.12 :)

As I said, one less DSHOT command.

> Can you point me to this 10x sending of dshots commands in the code? Can't seem to find it...

https://github.com/betaflight/betaflight/blob/b310f9b34873b5ef422152d88eb6a789af991538/src/main/drivers/dshot_command.c#L174

But note that changing this may result in the motor direction being inconsistently set and crash-flip mode not working - or failing to reset to normal after using crash-flip causing a flip-out when re-arming. It's also highly dependent on the ESC type and ESC firmware version as they have had different requirements over time. We don't not recommend changing this. The settings are conservative to handle all the different use-cases of different ESCs and firmware versions.

Lastly, your high speed camera timing based on the LED strip changing is somewhat flawed. The LED strip is only updated periodically on a schedule and is not realtime. So there's an inherent variability there. The default for the LEDSTRIP task is 100hz so 0.01s but it's a low-priority task so it will be more variable than that.

kevindehecker, 2019-08-12:
OK thanks for pointing to the repeat. I understand your warnings, but for the sake of gathering information I tried changing it anyway to ` repeat = 2`. Also decreased the following delays:
```
#define DSHOT_INITIAL_DELAY_US 1000
<..>
#define DSHOT_ESCINFO_DELAY_US 1200
#define DSHOT_BEEP_DELAY_US 1000
```

In good news, it was still flying. Bad news, the delay was not affected. Also considering your remark that all this should not affect spin up with motor_stop, I'm inclined to agree this must be esc related.

Still weird that reversing direction in 3d mode (which basically includes stopping) is so much faster though. I'll try digging a bit deeper in BLHeli then.

kevindehecker, 2019-09-12:
I had a look around in BLHeli, but assembly is not my cup of tea and it quickly took more time than I had. In any case I beleve this is not BF related. Thanks for the help!

closed 2019-09-12 by kevindehecker